motioneye-project/motioneyeos #2967: Raspberry Pi Zero W - Stuck On Rainbow Screen

DvKLM, 2022-12-19:
## Overview

When attempting to install motioneyeos **202006006** onto a **Raspberry Pi Zero W v1.0**, after first inserting the SD card into the RPi0, the Pi is stuck on the rainbow screen (remains there after a long time).

### Installation Details
 
- OS is Linux with Balena Etcher (same results when using `writeimage.sh` and `Raspberry Imager`)
- Attempting to install this version of motioneyeos: https://github.com/motioneye-project/motioneyeos/releases/download/20200606/motioneyeos-raspberrypi-20200606.img.xz 
- Tried several SD cards of various brands and sizes (8Gb - 128Gb, Sandisk).
- Used `unxz` to unzip the package prior to flashing.
- Accessing the RPi0 with monitor and keyboard so have not tried to connect to wireless.

### What Happens

- Insert SD card into RPi0W (with HDMI connector & USB keyboard)
- Insert Raspberry Pi Official power cable
- LED light does **not** turn on
- Remains stuck on rainbow screen

## Attempts to Resolve

After looking at previous issues, I found [this](https://github.com/motioneye-project/motioneyeos/issues/2419#issuecomment-645950906) post, suggesting that the Pi be updated.

- Flashed Raspberry Pi OS Lite 32 bit -> OS loaded fine
- Ran `sudo apt update && sudo apt full-upgrade -y`
- Attempted  `vcgencmd bootloader_version`, which returned `unknown`. Running `vcgencmd bootloader_config` shows as empty.
- Tried `sudo apt install rpi-eeprom` to resolve the above, but it was already installed and up-to-date (version 15.1-0).
- Tried to re-flash motioneyeos onto SD card anyway despite this, but still met with rainbow screen.

Any ideas greatly appreciated, thanks!

starbasessd, 2022-12-19:
The bootloader command is only for Pi4s.
The rpi-eeprom commands are only for Pi4s
Try using the [dev20201026](https://github.com/motioneye-project/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi-dev20201026.img.xz) version.
I use Win10, 7Zip, and Win32DiskImager to create motionEyeOS SDCards. RPiImager doesn't play well with the motionEyeOS images, especially if you have used the pre-sets in RPi-Imager at any time.
I use dd in linux and macos.

DvKLM, 2022-12-19:
Thanks so much.

I will try updating the bootloader via that link and see how it goes.

Just to say, I initially tried Etcher, and then writeimage.sh, and onlynused RPiImager as a last resort.

starbasessd, 2022-12-19:
You cannot update the bootloader on anything but the Pi4s, they are the only ones that have that ability.

DvKLM, 2022-12-19:
Apologies, I misunderstood, I thought the link was for the bootloader but i see its for a different motioneyeos image.

DvKLM, 2022-12-19:
dev20201026 flashed using both Etcher and `writeimage.sh`, but it still got stuck at the rainbow screen. However, this time the green LED came one as solid colour, which it did not do before. 

I also tried `dd`, as:

```
sudo dd if=motioneyeos-raspberrypi-dev20201026.img of=/dev/sdb
``` 

But this did the exact same as above. Rainbow screen with solid green LED.

starbasessd, 2022-12-19:
Yet if you do Raspberry OS, it works using either of those 2 methods (Not RPiImager)

DvKLM, 2022-12-19:
I installed RPi OS using RPImager.

I will try with Etcher or `dd` to install RPi OS and see if it works.

DvKLM, 2022-12-20:
I installed RPiOS successfully via Etcher, so I guess that isn't the issue?

starbasessd, 2022-12-20:
The only Mac I have runs High Sierra.
This doesn't like homebrew. You need homebrew or similar to install wget or xz utilities.
After digging for an hour or so, and many hoops found to jump through, I was able to get brew, wget, and xz utils to install.
I then used:
wget https://github.com/motioneye-project/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi-dev20201026.img.xz
to get the dev file for PiZero/Pi1
I then used :
xz -d ./motioneyeos-raspberrypi-dev20201026.img.xz
which decompressed to:
-rw-r--r--   1 kevin  staff  346030080 Dec  7  2021 motioneyeos-raspberrypi-dev20201026.img
I then attempted to dd the file to /dev/disk1 (my SDCard USB device) and it failed due to needing SIP disabled. I am stopping here for now. I will continue this afternoon...

starbasessd, 2022-12-20:
What bits don't you understand, as I need us to be on the same page.

DvKLM, 2022-12-20:
Just that I'm unsure what brew is, and also don't know what SIP is or how it relates.

starbasessd, 2022-12-20:
How did you install wget, xz? 
SIP is the security protocal that blocks access to certain terminal commands, files and folders, even with sudo.

DvKLM, 2022-12-20:
I didn't use wget, i just downloaded from the release page.

I installed `xz` via `sudo apt install xz-utils`

starbasessd, 2022-12-20:
Mac OS doesn't use apt...

DvKLM, 2022-12-20:
Ah, I'm on Linux. I did wonder why you mentioned macos earlier but I assumed you were working with what you had.

starbasessd, 2022-12-20:
I swear, I thought I saw you say you used Mac to create the SDCard. 
What is your setup for the machine you use to create the SDCard? X86-64 PC or ARM Pi? Which OS? I really want to try to reproduce your issue. I can fire up almost any OS, I have Ryzen and Intel, and can spin up almost anything...

DvKLM, 2022-12-20:
No worries, it happens! I'll try and describe everything which might be relevant.

- Running Pop!_OS 22.04 LTS
  - _Not sure if this would make a difference, but the OS is running from an external hard drive._
- Laptop with Intel i5 
- Balena Etcher is running as an appimage, from [here](https://github.com/balena-io/etcher/releases/download/v1.10.2/balenaEtcher-1.10.2-x64.AppImage?d_id=caf48ba1-c137-40d5-b48f-ba8d2a5b2dff&s_id=1671567044495).
- The RPi0W is not new. It was previously being used with PiHole, which ran ok.

DvKLM, 2023-02-03:
Hi, for some reason my account was banned! Managed to get it back. Sorry to drag this up again, did you have any luck in replicating the issue?

starbasessd, 2023-02-03:
No, I wasn't.
I used: 
Win 10, and Win32DiskImager
Win 10, and Balena Etcher
Win 10, WSL2 and Ubuntu & dd
Win 10, WSL2 and Debian & dd
Various versions of Debian (10 & 11) and dd on various machines (laptops and PCs)
Various versions of Ubuntu (20 & 22) and dd on various machines (laptops and PCs)
MacOS X High Sierra (10.13) and dd
RaspberryPiOS on Pi3 and Pi4 and dd
All writing [motionEyeOS from this source](https://github.com/motioneye-project/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi-dev20201026.img.xz) to several SDCards (different brands and sizes from 4GB to 64GB)
All with the same result: All good.
On the back of the PiZero, just under the GPIO, is a line of text. Can you please put it here? The whole line, including the FCC ID.

starbasessd, 2023-02-05:
@DvKLM 
Please, See previous comment https://github.com/motioneye-project/motioneyeos/issues/2967#issuecomment-1415121645 On the back of the PiZero, just under the GPIO, is a line of text. Can you please put it here? The whole line, including the FCC ID.

DvKLM, 2023-02-09:
Below is everything written below the pins.

FCC ID: 2ABCB-RPIZ2
IC: 20953-RPIZ2

DvKLM, 2023-02-09:
I'm coming to wonder if there is just something off with the board! If I could get hold of another zero I could try that too, but that's difficult. It is odd that it seems to run RasPiOS ok though...

starbasessd, 2023-02-09:
You don't have an RPi Zerw W, you have an RPi Zero2 wireless. It is NOT the same thing, The RPiZero2W has a different processor, and will NOT work with the RPiZero image.Use this one: https://github.com/motioneye-project/motioneyeos/issues/2875

DvKLM, 2023-02-09:
Problem in chair not in computer

This is very embarrassing! I will try this when next at home, but I imagine this will work. I'll update here when tried it. Sorry if this has been a massive waste of time.

starbasessd, 2023-02-09:
Not a waste of time if the lesson: check what hardware you have, is learned...

DvKLM, 2023-02-12:
It works! Thanks for all your help 🙏

On a separate note, I now can't get the camera module to show an image (vcgencmd returns that the camera is supported but not detected).

By all means close this issue and I can raise a new one if I can't manage to sort it out.

starbasessd, 2023-02-12:
If the camera was released after June of 2020, it's not installed in the kernel and probably won't work. If you need any of the HD CSI (ribbon cable) cameras, it's best to go with Raspbian Lite Buster, and install motionEye using the instructions [here](https://github.com/motioneye-project/motioneye/wiki/Install-On-Raspbian)

DvKLM, 2023-02-12:
Does that June 2020 rule apply to RiPi camera modules too? I kinda assumed that it would just work on plugging it in

starbasessd, 2023-02-12:
Yep, the reason 'it just works' is because the kernel gets updated by the RPi Foundation just before the camera comes out... Until/unless the devs update motionEyeOS, or you ['build your own update'](https://github.com/motioneye-project/motioneyeos/wiki/Building-From-Source) it won't get the latest kernel, and you'll need to run Buster plus motionEye...

DvKLM, 2023-02-12:
Cool, understood. Appreciate all the support with all this, was not expecting it but what you do is greatly appreciated 🙏