motioneye-project/motioneyeos #2968: Record issue

vertexgamer, 2022-12-21:
Your shit software won't record more than 1 file when a maximum file length is set. The homeassistant addon did work

coder0107git, 2022-12-21:
1) The MotionEye/OS projects are maintained and staffed by volunteers (the community).
2) The Home Assistant [add-on](https://community.home-assistant.io/t/home-assistant-community-add-on-motioneye/71826)/[integration](https://www.home-assistant.io/integrations/motioneye/) for MotionEye/OS are made by the [Home Assistant Community Add-ons Project](https://github.com/hassio-addons/addon-motioneye) and [Home Assistant Team](https://github.com/home-assistant/core/tree/dev/homeassistant/components/motioneye).
3) What do you mean by `The homeassistant addon did work`? (Please provide more details)
4) Since MotionEyeOS currently uses the old Python 2 version of MotionEye, it is very outdated. Until MotionEyeOS is updated it is better to install the dev version of MotionEye either as a docker container or on Raspberry Pi OS (Raspbian).
5) Does the problem appear on the latest (dev) version of MotionEye?

vertexgamer, 2022-12-22:
By **The homeassistant addon did work** i meant that the **maximum recording file length (in seconds)** works as intended on the addon, but on MotionEyeOs [20200606] it's broken and won't create a new file when the selected time is reached.
I haven't tested the latest dev version yet and probably won't do anytime soon

starbasessd, 2022-12-22:
dev20201026 of motionEyeOS is known to work correctly.

starbasessd, 2022-12-22:
Also, there are 3 primary types of postings, here.
1. People reporting an issue, either  bug or requesting info on how to do something.
2. People offering help.
3. Trolls with bad attitudes.

I had a tendency to remove postings by the third type, as they were non-helpful, refused to give info requested, and or were using HA or other 3rd party wrappers and blaming motionEye/motionEyeOS.  
The very first line indicates that the request for information was ignored, and is of the third variety. As  I am no longer a collaborator, I can only be the second option and ask the whole posting be removed, or at least closed and locked.

vertexgamer, 2022-12-22:
just tried your beloved dev20201026 for rpi3 and it's even more bugged than 20200606 release, as alongside many issues, the "apply" button is just missing and won't appear even after reboot. Honestly stop wasting my time.

closed 2022-12-22 by vertexgamer

starbasessd, 2022-12-22:
@vertexgamer You are wasting your own time. Thank you for closing.

starbasessd, 2022-12-22:
I have dev20201026 on a Pi3B+ and it works for me. what are your other settings? I have mine set to 30 seconds in motion detection mode, and these in Motion Detection:
[image]
After 32 seconds detection resets and allows the next recording. 
It is not designed as written to 'break up' files into whatever length you set in a motion triggered environment.