motioneye-project/motioneyeos #2969: vcgencmd get_camera showing not supported

SureshkumarSenthilkumar, 2023-01-05:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 4 B).

### Camera

I am using the following type of camera: CSI camera module(choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: Raspberry Pi camera Rev 1.3 (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: Internet(choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

starbasessd, 2023-01-05:
If you install RPiOS Buster (Legacy) on the SDCard, what does vcgencmd get_camera report?

SureshkumarSenthilkumar, 2023-01-06:
> If you install RPiOS Buster (Legacy) on the SDCard, what does vcgencmd get_camera report?

Supported=1, connected=1 when I connect csi camera onboard