motioneye-project/motioneyeos #2970: Camera modules v3?

Gonioul, 2023-01-10:
Are camera modules v3 supported?

alphanumeric007, 2023-01-11:
I don't see any new releases, so I would think, no not yet. The last official release was 2020. I have to use the Dev release on one or two of my Pi 4's to get Motion Eye up and running on those.

Gonioul, 2023-01-12:
Well, the wiki says any CSI camera _should_ be supported...

alphanumeric007, 2023-01-12:
I based my answer on what I'm seeing on the Pi forum. You might get video but I'm pretty sure the autofocus isn't going to work. Pi OS needed upgrades for the V3 which only fully works in libcamera. 
No auto focus is a no go for me.

Gonioul, 2023-01-12:
Well, for my case, if it's stuck on infinite, I don't care ^_^

alphanumeric007, 2023-01-13:
If you can get the full 12 mega pixel then your all set. I'm greedy, I want the auto focus to work. I have 8 Pi Camera setups here, all running Motion Eye OS. Two are Pan Tilt, auto focus on those would be awesome.

Gonioul, 2023-01-14:
I'm more interested in better low-light videos, no need for high quality which pi-zeros can't handle anyways...

alphanumeric007, 2023-01-15:
I gave up trying to use a Pi Zero, 3A's 3B's and 4B's here.

SheaStevenson, 2023-01-17:
I just plugged a v3 camera into my existing setup, and it doesn't work. But all that confirms is that it's not plug & play.

Camera: Pi Camera Module v3 (Regular)
Board: Pi Zero W
OS Version: 20200606

Mario-vminion, 2023-01-24:
is there another service what you can add to the pi to work with pi cam v3 instead of motioneyeos

starbasessd, 2023-01-24:
You casn use Raspbeian Buster Legacy and motionEye [Instructions here](https://github.com/motioneye-project/motioneye/wiki/Installation)

Mario-vminion, 2023-01-24:
is it supporting the v3

starbasessd, 2023-01-24:
Raspberry Pi OS Buster Legacy is put out by Raspberry Pi and should have PiCam V3 support in it. If not, then I doubt much of anything would.

Mario-vminion, 2023-01-24:
i wanted to use it for watching the 3d printer... from a location ..

starbasessd, 2023-01-24:
There are a number of packages out there that can stream video across the network using everything from PiZero to RP2040 to ESPx and more.

Gonioul, 2023-01-24:
If motioneyeos isn't going anywhere soon, I'd like to know the best alternate solutions, especially for pi zeros and fast network cameras.

kingkingyyk, 2023-01-27:
I've written a little [MJPEG server](https://github.com/kingkingyyk/Pi-MJPEG-Server
) for latest Raspbian that acts like a Fast Network Camera, based on the official PiCamera2 API sample `mjpeg_server.py` and `capture_mjpeg_v4l2.py`. 

Ideally you still need to run motioneye in another powerful Pi if you use Zero. ;)

starbasessd, 2023-03-30:
The issue is if you are using motionEyeOS, the camera V3 came out after it was compiled, and is not in the kernel. If you want to use a V3 camera, you need to use RaspiOS Buster Legacy and the [install instructions here](https://github.com/motioneye-project/motioneye/wiki/Install-On-Raspbian) Bullseye is broken for the needed modules for CSI cameras...