motioneye-project/motioneyeos #2971: Error installing on raspbian

Raspiguy, 2023-01-13:
I have a virtual machine running raspbian desktop. I have never had any issue installing a program because of this but when i try and run:

   cp /usr/local/share/motioneye/extra/motioneye.systemd-unit-local /etc/systemd/system/motioneye.service

it just outputs error:

cp: cannot stat '/usr/local/share/motioneye/extra/motioneye.systemd-unit-local': No such file or directory

Any ideas?

starbasessd, 2023-01-13:
Use the install instructions for Debian, not Raspbian.

Raspiguy, 2023-01-14:
> Use the install instructions for Debian, not Raspbian.

Same error running 

  cp /usr/local/share/motioneye/extra/motioneye.systemd-unit-local /etc/systemd/system/motioneye.service

reference https://github.com/motioneye-project/motioneye/wiki/Install-on-Debian-11-Bullseye