motioneye-project/motioneyeos #2972: rebooting after brief connection.

JohnWiik, 2023-01-16:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motioneyeos-raspberrypi-dev20201026.

### Board Model

I am using the following board/model: Raspberry PI Zero W.

### Camera

I am using the following type of camera: Raspberry Pi NoIR Camera Module V2 - 8MP 1080P30.

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

**The problem I am having right now:**
Everything worked fine for a little bit, then after a few seconds, the system rooted itself again. 
The code in my wpa_supplicant.conf file is below:
`country=US
update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
	scan_ssid=1
        ssid="xxxxxx"
        psk="yyyyyy"
}`

Can someone help me resolve this issue? How can I stop it from rebooting after the browser has successfully connected to the IP address. 

https://user-images.githubusercontent.com/62785162/212583216-77e9b66a-4ef8-483f-ba9c-196fb0f2b276.mp4

JohnWiik, 2023-01-18:
@starbasessd hi, sorry to bother you, but I wonder if you could help me with this issue? Thank you very much in advance!

starbasessd, 2023-01-18:
Please post the motioneye.log & messages.log. If they are large, you can ssh or console in and run:
logrotate -f /wtc/logrotate.conf 
and then post the logs after the next reboot
You can also post the results of the command
vcgencmd get_throttled

Gonioul, 2023-01-24:
I'd guess it's either a low power problem or a high thermal problem...