motioneye-project/motioneyeos #2973: Motioneye Resolution won't change

Phlupke15, 2023-01-19:
Hey everyone, 
I am running motioneye on a Ubuntu proxmox Server using an usb Webcam (E-meet c960). It's Working good, but the resolution is stuck at 480p. 

I can change the resolution in motioneye Webbrowse, but  the video resolution won't change. 

I tried change the motion config file and changed the resolution to 1920x1080 and 15 FPS. But when I change it in the browser nothing changes. 

What can I try or how can I fix the problem. 

Phil

Phlupke15, 2023-01-19:
Hey, im running it direct on a proxmox server. Proxmox is installed on a Wyse5070 with 8GB.
Its running

motionEye Version | 0.42.1
-- | --
Motion Version | 4.2.2
OS Version | Ubuntu 20.04

The Webcam is on USB2.0.

starbasessd, 2023-01-19:
If you were on Debian, I'd suggest installing USBTOP to see bandwidth usage. I'd bet it's too much at the 1920x1080. Can you try 1280x720, or slightly lower?

Phlupke15, 2023-01-20:
How do I use USBTOP? 

I only can Change it to lower resolutions then 680x480.

It is running good, also at higher FPS but the resolution won't change to higher than that. 

Its the only camera and the proxmox Server got 2 cores and 2GB RAM. Should I try more ram?

starbasessd, 2023-01-20:
Under Video Device, Video Resolution, select Custom, and enter 1280x720. 
BTW, that is a low spec system to be running ProxMox on, especially if you are running any other virtual machines on it. MotionEye is very CPU, RAM, and other resources hungry. I run my secondary motionEye Server on a VBox VM (minimal Debian 11) with 4 cores and  4GB of RAM, supporting 5 IP cameras at an average of 1280x720 and 10 FPS. The overall server can tap 32GB RAM and 16 cores.