motioneye-project/motioneyeos #2975: RTSP streams not in real time

DJAlexOfficiel, 2023-01-22:
Hello,

When I use my RTSP stream on VLC the stream is in real time, but when I put it on motion eye the stream is not real time I have 5 seconds of delay. 

I would like to understand why?

Thanks in advance.

starbasessd, 2023-01-22:
Some of the 'delay' is caused by the processing of the video stream to detect motion. The higher the resolution used, the more processing needed, and depending on the SBC used, it can be significant.

DJAlexOfficiel, 2023-01-22:
> Some of the 'delay' is caused by the processing of the video stream to detect motion. The higher the resolution used, the more processing needed, and depending on the SBC used, it can be significant.

I have a resolution of 600x800 at 15fps without motion detection I have disabled everything but it still causes me a delay but that is huge 5 sec. How can I fix this problem? Do you have an idea?

starbasessd, 2023-01-22:
What are you running on? PC? Pi? Something else?
I run 2 RTSP cameras, and 3 mjpeg cameras, and a USB Webcam, at an average of 1280x720 10FPS and do not see any delay other than occasional < 1 second type when all cameras have a lot of motion. This is on an i5-1.9GHz processor, running Debian 11, with 6GB RAM...

DJAlexOfficiel, 2023-01-22:
> What are you running on? PC? Pi? Something else?
> I run 2 RTSP cameras, and 3 mjpeg cameras, and a USB Webcam, at an average of 1280x720 10FPS and do not see any delay other than occasional < 1 [...]

I am under raspberry pi 3 it is small the problem? After trying to put under a raspberry pi 4 8go of ram I have a problem the soft motion eye does not work my raspberry pi 4 after 15min this one still does not connect to my box internet even if I am in cable rj45
Traduit avec DeepL https://www.deepl.com/app/?utm_source=android&utm_medium=app&utm_campaign=share-translation

starbasessd, 2023-01-22:
There are several threads here about issues with Pi4s and the current (dev20201026) version of motionEyeOS. Yes, the Pi3 is underpowered. If you have boot issues on the Pi4 and can't find the resolution here, I would install Raspbian Buster Legacy (preferred) or Bullseye (both Lite versions) and follow the appropriate install instructions [here](https://github.com/motioneye-project/motioneye/wiki/Installation)

DJAlexOfficiel, 2023-01-22:
> There are several threads here about issues with Pi4s and the current (dev20201026) version of motionEyeOS. Yes, the Pi3 is underpowered. If you have boot issues on the Pi4 and can't find the [...]

I'll try tomorrow but with a raspberry pi 4 would solve the problem?

starbasessd, 2023-01-22:
No guarantee, but Should be better.

DJAlexOfficiel, 2023-01-22:
> No guarantee, but Should be better. 

Otherwise you have to switch to an i3 or i5 style computer?

DJAlexOfficiel, 2023-01-23:
> No guarantee, but Should be better. 

I tested on an i5 8go of ram that works fine so I will ch aher the config to run motion eye I thank you for your involvement.

closed 2023-01-25 by DJAlexOfficiel