motioneye-project/motioneyeos #2976: Willing to Maintain HAOS Addon?

Ryu945, 2023-01-28:
I heard this project was taken over by a new maintainer.   Would it be possible for you to update the addon for HAOS and continue maintaining it since the old project is abandoned?  It is so out of date that sftp doesn't even work on it anymore.

starbasessd, 2023-01-28:
Are you asking for someone to do the HA plugin, to do the motionEye update or what? They are 2 different projects (3 if you include motionEyeOS).
The HA plugin was never really well supported, always blaming issues on motionEye, but if motionEye or motionEyeOS was installed as a separate entity, it didn't exhibit the issues claimed. 
MotionEyeOS is stand-alone and not used/usable in HA.
The new motionEye Devs are working on getting it switched off of Python2.7 to Python3.x and are not actively working on bringing up motionEyeOS until they complete the conversion.

Ryu945, 2023-01-29:
Thankyou for the information.   After the migration, could you update the HA addon?

coder0107git, 2023-01-29:
@Ryu945 I think there has been a misunderstanding. Both the `motionEye` and `motionEyeOS` projects are housed and maintained by the [`motioneye-project` github organization](https://github.com/motioneye-project). The HA addon isn't advocated for (officially), supported, or maintained by the `motionEye` devs. Since the HA addon is maintained by a different organization they will have to update it when `motionEye 0.43` is released.

Ryu945, 2023-01-30:
Thankyou for the information.