motioneye-project/motioneyeos #2978: Raspberry Pi 4 reboot cycle seconds after boot up

frakman1, 2023-02-16:
My setup used to work for years before finally failing due to constant reboots while booting up. 
I am using a 5V/2.5A power supply/adapter from Canakit so it's not voltage related. 
I don't see any obvious error messages in the bootup logs (included below).

### Preliminary Docs

- [x] I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

- [x] I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: Not sure since it doesn't fully boot. Looks like:
`[    0.000000] Linux version 4.14.30-v7 (root@042a70bf8ec8) (gcc version 6.4.1 20170707 (Linaro GCC 6.4-2017.08)) #1 SMP Thu Jun 28 20:57:53 UTC 2018`

`[    0.070083] raspberrypi-firmware soc:firmware: Attached to firmware from 2018-03-23 16:35`

### Board Model

I am using the following board/model:  Raspberry PI 3B+ (v1.2)

### Camera

I am using the following type of camera: ribbon cable to onboard CSI port

My camera model is: Raspberry Pi Camera v1.3 

### Network Connection

My motionEyeOS unit is connected to the network via: hard wired

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

None

### Log Files

I consider the following log files relevant to this issue:

Serial console boot log:

```
[   20.341487] reboot: Restarting system
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.30-v7 (root@042a70bf8ec8) (gcc version 6.4.1 20170707 (Linaro GCC 6.4-2017.08)) #1 SMP Thu Jun 28 20:57:53 UTC 2018
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 8 MiB at 0x37800000
[    0.000000] On node 0 totalpages: 229376
[    0.000000] free_area_init_node: node 0, pgdat 80c7ef00, node_mem_map b7016000
[    0.000000]   Normal zone: 2016 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 229376 pages, LIFO batch:31
[    0.000000] random: fast init done
[    0.000000] percpu: Embedded 17 pages/cpu @b6fbf000 s38604 r8192 d22836 u69632
[    0.000000] pcpu-alloc: s38604 r8192 d22836 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 227360
[    0.000000] Kernel command line: 8250.nr_uarts=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.fiq_fix_enable=1 console=ttyS0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait ro rootflags=noload panic=10 loglevel=8 ipv6.disable=1
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 887480K/917504K available (7168K kernel code, 550K rwdata, 2056K rodata, 1024K init, 706K bss, 21832K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[... 208 lines cut ...]
[    3.708305] snd_bcm2835: module is from the staging directory, the quality is unknown, you have been warned.
[    3.723921] bcm2835_alsa bcm2835_alsa: card created with 8 channels
[    3.862576] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[    3.874008] brcmfmac: brcmf_fw_map_chip_to_name: using brcm/brcmfmac43430-sdio.bin for chip 0x00a9a6(43430) rev 0x000001
[    3.888669] usbcore: registered new interface driver brcmfmac
[    4.112205] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: Aug 29 2016 20:48:16 version 7.45.41.26 (r640327) FWID 01-4527cfab
[    4.129905] brcmfmac: brcmf_c_preinit_dcmds: CLM version = API: 12.2 Data: 7.11.15 Compiler: 1.24.2 ClmImport: 1.24.1 Creation: 2014-05-26 10:53:55 Inc Data: 9.6.3 Inc Compiler: 1.29.4 Inc ClmImport: 1.31.4 Creation: 2016-08-29 20:46:38 
[   20.191489] reboot: Restarting system
```
 
cmdline.txt:
```
dwc_otg.fiq_fix_enable=1 console=serial0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait ro rootflags=noload panic=10 loglevel=8 ipv6.disable=1
```

config.txt:
```
gpu_mem=128
dtparam=i2c_arm=on
dtparam=i2s=on
dtparam=spi=on
dtparam=audio=on
disable_camera_led=1
enable_uart=1
```

frakman1, 2023-02-16:
Thank you for your response. 
Nothing changed. It just stopped working one day and wouldn't show up on the network, I had to enable logs on the flash drive files before I could see the reboot cycle logs.
I have no USB devices plugged in at all. It's just got a network cable hooked up and the CSI camera. Now, not even the network cable is connected while I troubleshoot. I even disconnected the camera for good measure but still the same behaviour.
It's in a reboot cycle so unfortunately no interactive prompt and no access to any files.

starbasessd, 2023-02-16:
Do you have access to a Linux PC to read the 3rd partition?

frakman1, 2023-02-16:
Oh you mean mounting the flash drive and reading the ext4 partitions? 
Yes, I think I can do that on my Mac.

starbasessd, 2023-02-16:
Let me know.

frakman1, 2023-02-16:
I was (finally) able to mount and read the ext4 partition on my Mac.
The messages file is 3.5GB and the last 5000 line or so just look like this:
```
Feb 12 14:18:30 piCam1 user.debug motion: [1:ml1] [DBG] [VID] v4l2_capture: the_buffer index 1 Address (73406000)
Feb 12 14:18:30 piCam1 user.debug motion: [1:ml1] [DBG] [VID] v4l2_capture: 1) vid_source->pframe 1
Feb 12 14:18:30 piCam1 user.debug motion: [1:ml1] [DBG] [VID] v4l2_capture: 2) vid_source->pframe 1
Feb 12 14:18:30 piCam1 user.debug motion: [1:ml1] [DBG] [VID] v4l2_capture: 3) vid_source->pframe 2 vid_source->buf.index 2
Feb 12 14:18:30 piCam1 user.debug motion: [1:ml1] [DBG] [VID] v4l2_capture: vid_source->buf.bytesused 3110400
Feb 12 14:18:30 piCam1 user.debug motion: [1:ml1] [DBG] [VID] v4l2_capture: the_buffer index 2 Address (73109000)
Feb 12 14:18:30 piCam1 user.debug motion: [1:ml1] [DBG] [VID] v4l2_capture: 1) vid_source->pframe 2
Feb 12 14:18:30 piCam1 user.debug motion: [1:ml1] [DBG] [VID] v4l2_capture: 2) vid_source->pframe 2
Feb 12 14:18:30 piCam1 user.debug motion: [1:ml1] [DBG] [VID] v4l2_capture: 3) vid_source->pframe 3 vid_source->buf.index 3
Feb 12 14:18:30 piCam1 user.debug motion: [1:ml1] [DBG] [VID] v4l2_capture: vid_source->buf.bytesused 3110400
Feb 12 14:18:30 piCam1 user.debug motion: [1:ml1] [DBG] [VID] v4l2_capture: the_buffer index 3 Address (72e0c000)
Feb 12 14:18:30 piCam1 user.debug motion: [1:ml1] [DBG] [VID] v4l2_capture: 1) vid_source->pframe 3
Feb 12 14:18:30 piCam1 user.debug motion: [1:ml1] [DBG] [VID] v4l2_capture: 2) vid_source->pframe 3
Feb 12 14:18:30 piCam1 user.debug motion: [1:ml1] [DBG] [VID] v4l2_capture: 3) vid_source->pframe 0 vid_source->buf.index 0
Feb 12 14:18:30 piCam1 user.debug motion: [1:ml1] [DBG] [VID] v4l2_capture: vid_source->buf.bytesused 3110400
Feb 12 14:18:30 piCam1 user.debug motion: [1:ml1] [DBG] [VID] v4l2_capture: the_buffer index 0 Address
```
`dmesg.log` shows the last few hundred lines look like this:

```
---- booting motionEyeOS 20180627 ----
[Wed Dec 31 18:59:59 1969] Booting Linux on physical CPU 0x0
[Wed Dec 31 18:59:59 1969] Linux version 4.14.30-v7 (root@042a70bf8ec8) (gcc version 6.4.1 20170707 (Linaro GCC 6.4-2017.08)) #1 SMP Thu Jun 28 20:57:53 UTC 2018
[Wed Dec 31 18:59:59 1969] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[Wed Dec 31 18:59:59 1969] CPU: div instructions available: patching division code
[Wed Dec 31 18:59:59 1969] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[Wed Dec 31 18:59:59 1969] OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
[Wed Dec 31 18:59:59 1969] Memory policy: Data cache writealloc
[Wed Dec 31 18:59:59 1969] cma: Reserved 8 MiB at 0x37800000
[Wed Dec 31 18:59:59 1969] On node 0 totalpages: 229376
[Wed Dec 31 18:59:59 1969] free_area_init_node: node 0, pgdat 80c7ef00, node_mem_map b7016000
[Wed Dec 31 18:59:59 1969]   Normal zone: 2016 pages used for memmap
[Wed Dec 31 18:59:59 1969]   Normal zone: 0 pages reserved
[Wed Dec 31 18:59:59 1969]   Normal zone: 229376 pages, LIFO batch:31
[Wed Dec 31 18:59:59 1969] random: fast init done
[Wed Dec 31 18:59:59 1969] percpu: Embedded 17 pages/cpu @b6fbf000 s38604 r8192 d22836 u69632
[Wed Dec 31 18:59:59 1969] pcpu-alloc: s38604 r8192 d22836 u69632 alloc=17*4096
[Wed Dec 31 18:59:59 1969] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[Wed Dec 31 18:59:59 1969] Built 1 zonelists, mobility grouping on.  Total pages: 227360
[Wed Dec 31 18:59:59 1969] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.fiq_fix_enable=1 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait ro rootflags=noload panic=10 quiet loglevel=1 ipv6.disable=1
[Wed Dec 31 18:59:59 1969] PID hash table entries: 4096 (order: 2, 16384 bytes)
[Wed Dec 31 18:59:59 1969] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[Wed Dec 31 18:59:59 1969] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[Wed Dec 31 18:59:59 1969] Memory: 887480K/917504K available (7168K kernel code, 550K rwdata, 2056K rodata, 1024K init, 706K bss, 21832K reserved, 8192K cma-reserved)
[Wed Dec 31 18:59:59 1969] Virtual kernel memory layout:
[Wed Dec 31 18:59:59 1969] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[Wed Dec 31 18:59:59 1969] ftrace: allocating 24911 entries in 74 pages
[Wed Dec 31 18:59:59 1969] Hierarchical RCU implementation.
[Wed Dec 31 18:59:59 1969] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[Wed Dec 31 18:59:59 1969] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[Wed Dec 31 18:59:59 1969] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[Wed Dec 31 18:59:59 1969] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[Wed Dec 31 18:59:59 1969] Switching to timer-based delay loop, resolution 52ns
[Wed Dec 31 18:59:59 1969] Console: colour dummy device 80x30
[Wed Dec 31 18:59:59 1969] console [tty1] enabled
[Wed Dec 31 18:59:59 1969] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[Wed Dec 31 18:59:59 1969] pid_max: default: 32768 minimum: 301
[Wed Dec 31 18:59:59 1969] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[Wed Dec 31 18:59:59 1969] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[Wed Dec 31 18:59:59 1969] Disabling memory control group subsystem
[... cut ...]
[Wed Dec 31 18:59:59 1969] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.fiq_fix_enable=1 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait ro rootflags=noload panic=10 quiet loglevel=1 ipv6.disable=1
[Wed Dec 31 18:59:59 1969] PID hash table entries: 4096 (order: 2, 16384 bytes)
[Wed Dec 31 18:59:59 1969] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[Wed Dec 31 18:59:59 1969] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[Wed Dec 31 18:59:59 1969] Memory: 887480K/917504K
```

starbasessd, 2023-02-16:
Out of curiosity how much free space is available on each of the 3 partitions?
Can you also post the /data/log/boot.log please?

frakman1, 2023-02-16:
Well, it looked like the main partion was almost full so I deleted the massive messages file and some media files for good measure.

It didn't make a difference and it keeps rebooting at the same place.
It's strange that there are no errors in the logs.

```
Filesystem                Size   Used  Avail Capacity iused      ifree %iused  Mounted on
msdos://disk4s1/NO NAME   20Mi  8.8Mi   11Mi    44%       1          0  100%   /Volumes/NO NAME
/dev/disk4s3              30Gi   11Gi   18Gi    38%    1949    1935763    0%   /Volumes/Untitled 1
/dev/disk4s2             180Mi  173Mi  7.4Mi    96%    7223      38961   16%   /Volumes/Untitled
```

Showing the last few iterations of boot.log:
```
---- booting motionEyeOS 20180627 ----
 * Detecting disk device: /dev/mmcblk0
 * Mounting filesystems: done
 * Remounting boot partition read-write: done
 * Remounting root partition read-write: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Configuring CPU frequency: done
 * Starting throttle watcher: done
 * Starting eudev: done
 * Starting watchdog: done
 * Configuring wired network: dhcp
 * Setting current date using http: Thu Feb 9 00:00:18 EST 2023
 * Starting http date updater: done
 * Starting crond: done
 * Starting sshd: done
 * Starting proftpd: done
 * Starting smbd: done
 * Star---- booting motionEyeOS 20180627 ----
 * Detecting disk device: /dev/mmcblk0
 * Mounting filesystems: done
 * Remounting boot partition read-write: done
 * Remounting root partition read-write: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Configuring CPU frequency: done
 * Starting throttle watcher: done
 * Starting eudev: done
 * Starting watchdog: done
 * Configuring wired network: dhcp
 * Setting current date using http: Fri Feb 10 00:00:22 EST 2023
 * Starting http date updater: done
 * Starting crond: done
 * Starting sshd: done
 * Starting proftpd: done
 * Starting smbd: done
 * Starting nmbd: done
 * Starting motioneye: done
 * Executing user init script: Starting ngrok-eye
Starting ngrok-eye2
Starting tailscale
done
 # Interface eth0 has IP address 10.0.0.205/24
 # Default gateway is 10.0.0.1
 # DNS server address is 69.252.80.80
---- shutting down motionEyeOS 20180627 ----
 * Stopping motioneye: done
 * Stopping smbd: done
 * Stopping nmbd: done
 * Stopping proftpd: done
 * Stopping sshd: done
 * Stopping crond: done
 * Stopping date updater: done
 * Stopping network: done
[... 203 lines cut ...]
 * Starting throttle watcher: done
 * Starting eudev: done
 * Starting watchdog: done
 * Configuring wired network: no link
---- shutting down motionEyeOS 20180627 ----
 * Stopping motioneye: failed
 * Stopping smbd: failed
 * Stopping nmbd: failed
 * Stopping proftpd: failed
 * Stopping sshd: failed
 * Stopping crond: failed
 * Stopping date updater: done
 * Stopping network: done
 * Stopping eudev: done
 * Stopping throttle watcher: done
 * Stopping syslogd: done
 * Detecting disk device: /dev/mmcblk0
---- booting motionEyeOS 20180627 ----
 * Detecting disk device: /dev/mmcblk0
 * Mounting filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Configuring CPU frequency: done
 ```

frakman1, 2023-02-16:
One of the last things I did was add a nightly reboot via `/etc/crontabs/root`:
```
0 0 * * * /sbin/reboot
```

I thought that might be kicking in due to a race condition before it gets system time and thinks it has to reboot right away because it thinks it's midnight.

I tried deleting the file but it comes right back.
I modified it to `10 10 * * * /sbin/reboot` but no change in behavior (confirmed file was updated)
I deleted the line but also no change in behavior  (confirmed file was updated)

starbasessd, 2023-02-16:
From the boot.log:
>---- booting motionEyeOS 20180627 ----
 * Detecting disk device: /dev/mmcblk0
 * Mounting filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Configuring CPU frequency: done
 * Starting throttle watcher: done
 * Starting eudev: done
 * Starting watchdog: done
 * Configuring wired network: no link
---- shutting down motionEyeOS 20180627 ----

A this is a Very old version, which the Pi4 didn't exist then, so I don't know where you got it.
B It is not seeing a link for the Ethernet, or WiFi, which will cause a panic reboot in all models. 
C The first version supporting the Pi4 was 20190904, but if you are going to upgrade, I would go with dev20201026.

frakman1, 2023-02-16:
A This is not running on a RPi4. It's a RP3+ as indicated in the Board Model section above.
A I got it from the release page. It's an official image.
B OK, by the time I got the serial console on, I had disconnected the network cable. It wasn't working even with the network cable connected so I will try reconnecting it and looking at the logs again.
C I was planning on upgrading to the latest since I am in the best position to do so now but I want to get to the bottom of this first otherwise I might just get the same error on the new one and lose all my customizations/configurations.

starbasessd, 2023-02-16:
A your title says Pi4...
B Look at the LEDs for the Ethernet port on the Pi. If they aren't lighting up, try a diferent port on the switch/router, or try another known good Ethernet cable.
C for the Pi3B+ be sure to use [this download](https://github.com/motioneye-project/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi3-dev20201026.img.xz)

frakman1, 2023-02-16:
Oops! You're right. Sorry about that. Must have been a typo. Will fix it...
Both sides had blinking lights when plugged in. I tried various routers and switches.
I'll report back when I have more data.

frakman1, 2023-02-17:
OK, I put some text at the end of `boot.log`, connected the network cable (LEDs blinking) and I still get the same behavior.
I don't understand how not having a network link should cause a kernel panic and reboot. I don'tknow of any system that reboots because it doesn't have a working network connection. Is this specific to motioneyeos?

boot.log:
```
//////////////////////////////////////////////////////////////////////////////////////////////////
START
//////////////////////////////////////////////////////////////////////////////////////////////////
---- booting motionEyeOS 20180627 ----
 * Detecting disk device: /dev/mmcblk0
 * Mounting filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Configuring CPU frequency: done
 * Starting throttle watcher: done
 * Starting eudev: done
 * Starting watchdog: done
 * Configuring wired network: no link
---- shutting down motionEyeOS 20180627 ----
 * Stopping motioneye: failed
 * Stopping smbd: failed
 * Stopping nmbd: failed
 * Stopping proftpd: failed
 * Stopping sshd: failed
 * Stopping crond: failed
 * Stopping date updater: done
 * Stopping network: done
 * Stopping eudev: done
 * Stopping throttle watcher: done
 * Stopping syslogd: done
 * Detecting disk device: /dev/mmcblk0
```

frakman1, 2023-02-17:
Just as an FYI, I installed the latest release on a different SD card and ran it on the same RPI3B+ it comes up all the way and lets me login via the webUI.
This tells me there's nothing wrong with the hardware or network cables etc.

Any troubleshooting ideas are welcome, otherwise, I might just start over again from scratch with the new one.

frakman1, 2023-02-17:
Of course, now I find out that Google Drive uploads don't work with the new version. Another reason I don't like to upgrade unecessarily. I see that there is an [issue](https://github.com/motioneye-project/motioneyeos/issues/2954) tracking that and that it is not resolved because Google shutdown access apps using older authentication/login methods and motioneyos hasn't updated the login logic.

starbasessd, 2023-02-17:
Yes the panic reboot is specific to motionEyeOS intentionally, as it is written for headless use, and if there isn't a network, and it is headless, then there is no way to access it.
Your message https://github.com/motioneye-project/motioneyeos/issues/2978#issuecomment-1433928246 indicates that the original SD Card has some corruption in the system. Since there are many upgrades between the version that has issues, and the latest, you wouldn't be able to do a simple backup/restore anyway, re-creating your camera settings is going to be the best option. You MIGHT be able to re-load the old version, but I doubt GDrive will work even with the old key as the changes made were on Google's side. If you were to attempt it, I would copy all the motion and motioneye and camera files from /data/etc/ on the old SD Card and put them in the new card... (you have to mount the partition as R/W in your Mac to do so, I believe)...

frakman1, 2023-02-17:
Thank you for your insight. You may be right about the old SD card corruption although it appears to be functional (read and writeable, folders intact)  but there's no telling if some system file is corrupt or not.

Unfortunately, part of Google's troubleshooting instructions involved deleting the permitted third party app on Google's security page and re-adding it, only to later find that re-adding is not possible (!!) Unfortunately, no reversion is possible now (with backed up keys/files) even though it was working the last time I checked about a week ago. Oh well. I hope motioneyeos updates Google Drive integration because that was the only and most convenient way for me to view the recordings remotely and in a browser without any fuss.

starbasessd, 2023-02-17:
I don't know how bad you want to save on GDrive, but...
You can get the GDrive mounted on Windows and Mac (and Linux) and share the folder to motionEyeOS (I've done it recently on a Debian install of motionEye and had GDrive mounted on it, and used the folder...

frakman1, 2023-02-17:
Nice thought. Can I install `rclone` on the motionEyeOS directly and access it that way?

starbasessd, 2023-02-17:
Unfortunately, not easily. dpkg, and apt are not available in motionEyeOS. If files can be copied to the correct folders, maybe, but that is WAY beyond any support I provide here.

frakman1, 2023-02-18:
Out of curiousity:
- How do I disable the 'reboot when no network detected' logic?
- How do I get the serial console to be interactive and allow me to login. I would like to troubleshoot directly via serial console without the need for network access.
 Currently it stops logging when it gets network access around here and does not respond to user keyboard input:
```
    7.359066] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xCDE1
[    8.524611] IPv6: Loaded, but administratively disabled, reboot required to enable
[    9.055255] system date is 2023-02-18 08:50:10
```

starbasessd, 2023-02-18:
Here are several links to adjust settings:
[Watchdog settings](https://github.com/ccrisan/thingos/wiki/watchdog.conf)
[os startup](https://github.com/ccrisan/thingos/wiki/os.conf) is the location for disabling network services if no network: OS_NETWORKLESS, but this disables any service that requires/uses networking.

Setting up serial console should be similar to any other Raspbian Buster with the exception of some commands are missing. 

What is your goal? If you plug in a monitor and keyboard, it works normally, with read access to /boot & /, and full access to /data

starbasessd, 2023-02-18:
The primary description of the config files in the last 2-3 versions is here:[Configuration](https://github.com/motioneye-project/motioneyeos/wiki/Configuration)

frakman1, 2023-02-18:
Thanks for the network related links.

As for serial, I enabled the UART via the changes to the`cmdline.txt` and `config.txt` files described above.

I hooked up a USB-to-serial console cable (like [this](https://www.adafruit.com/product/954) )to the UART ports of the Pi and connected to my laptop and am able to get serial console logs but they are non-interactive and stops logging after a brief blurb at bootup (stops after network access)

Normally, I am able to login and access the Linux # console via serial console ports connected to other Raspberry Pis with just those changes. MotionEyeOS seems to have a different serial console configuration unless I am doing something wrong. The only way to get to the interactive console now is to ssh in. My goal is to do this over serial console so I can troubleshoot the original problem with the older software on the other SD card. Currently, this is not possible since it keeps rebooting and I have to keep taking out the SD card, mount it on my laptop and view the boot.log file for clues.

frakman1, 2023-02-18:
In other news, I was able to get rclone to work using a helpful [motioneyos extension project ](https://github.com/goldfix/motioneyeos_ext) but had to update the rclone binary to 1.60.1 first.

starbasessd, 2023-02-18:
> In other news, I was able to get rclone to work using a helpful [motioneyos extension project ](https://github.com/goldfix/motioneyeos_ext) but had to update the rclone binary to 1.60.1 first.

Very cool. Didn't know this existed. Will try it soon...

starbasessd, 2023-02-18:
> Thanks for the network related links.
>
> As for serial, I enabled the UART via the changes to the`cmdline.txt` and `config.txt` files described above.
>
> I hooked up a USB-to-serial console cable (like [...]

Since motionEyeOS is not running an a 'real' Debian or Raspberry Pi OS, I do not know what alteration(s) were made to the kernel and support files in BuildRoot. The instructions are there to [BYO](https://github.com/motioneye-project/motioneyeos/wiki/Building-From-Source) but even they are 2+ years out of date, and RPiFoundation has made major changes as well, in both Buster and Bullseye... I am a masochist and support the Last Know Good because I believe(d) in it and still use it myself, and couldn't pay Calin money, but helped support his software as that is where my skill set lies...

frakman1, 2023-02-18:
> > In other news, I was able to get rclone to work using a helpful [motioneyos extension project ](https://github.com/goldfix/motioneyeos_ext) but had to update the rclone binary to 1.60.1 first.
> 
> Very cool. Didn't know this existed. Will try it soon...

Yup, instead of just doing a cronjob, I just put the rclone command directly into the WebUI field of command to run when media file is created:

[image]

Hopefully, this functionality might one day be built in to motioneyeos. I created an app in Google Console with it's own client id/secret for better performance. Using default rclone settings uses a shared service that everyone uses that is rate limited.

My configs for reference:

`/usr/bin/rclone.config`
```
[GoogleDrive]
type = drive
client_id = xxxxapps.googleusercontent.com
client_secret = xxxx
scope = drive
root_folder_id =
team_drive =
token = {"access_token":"xxx","token_type":"Bearer","refresh_token":"xxxx","expiry":"2023-02-18T11:56:30.616837082-05:00"}
```
`/usr/bin/rclone_local.config`
```
CAM_FOLDER=/data/output/Camera1
REMOTE_FOLDER=/PiCam1
```
Command to copy to Google Drive:
`/usr/bin/rclone_tool.sh run copy GoogleDrive`

Troubleshoot:
`/var/log/rclone.log`

frakman1, 2023-02-18:
> Here are several links to adjust settings:
> [Watchdog settings](https://github.com/ccrisan/thingos/wiki/watchdog.conf)
> [os startup](https://github.com/ccrisan/thingos/wiki/os.conf) is the location [...]

I don't see any `watchdog.conf` file, only `watch.conf` that seems use `LINK_WATCH` for something related.
I can't follow the logic but I think I have to somehow set `${panic}` to `false` somewhere in [here](https://github.com/ccrisan/thingos/blob/dd274f6f6ec46b8f392e50a83842f0fd9a682ed8/board/common/overlay/etc/init.d/S40network) if I want it to not freak out if there's no network connection. 

Or maybe I can just set 
`LINK_WATCH="false"` in `/data/etc/watch.conf` instead. Will try it later...