motioneye-project/motioneyeos #2981: Raspberry Pi Camera Module 3 Not Recognized in Motion OS

SierraArgo, 2023-03-10:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606

### Board Model

I am using the following board/model: Raspberry Pi Zero W

### Camera

I am using the following type of camera: Pi Camera (MMAL?)

My camera model is: RPi Camera Module 3

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

N/A

### Log Files

I consider the following log files relevant to this issue:

Available on request

starbasessd, 2023-03-10:
motionEyeOS 20200606 & dev20201026 were released before the driver for PiCam3 were in the kernel. If you need to use a PiCam3, you will need to use RaspberryPiOS Legacy (Buster) with motionEye. [Install instructions here](https://github.com/motioneye-project/motioneye/wiki/Installation) on the right column, until the new devs get around to fixing and updating motionEyeOS.

SierraArgo, 2023-03-12:
Ok cool, so I assume a Pi Cam 2 would work? I had one, but it seems to be busted so I grabbed a 3 lol.

starbasessd, 2023-03-12:
[Release date of PiCam2](https://www.raspberrypi.com/products/camera-module-v2/) Yes, it works with the 20200606 and dev20201026 releases.

SierraArgo, 2023-03-12:
Ok, running into some new issues now getting it to see my OMV NAS. Any logs I can send over for troubleshooting? I am getting the following error: "Accessing network share failed: cannot mount network share!"

Not sure if it's PEBKAC or not lol.

starbasessd, 2023-03-12:
What kind of network share is available on the OMV NAS? It doesn't appear to support FTP without a plugin. 
If you are using 'Upload Service' what do you have selected?

SierraArgo, 2023-03-12:
I had intended to use the network share option since OMV supports SMB. Would that not work?

starbasessd, 2023-03-13:
Easily with motionEye on another OS. Little more difficult with motionEyeOS
[image]
Set Storage Device to Network Share
Set Network Server to the IP Address of the OMV server
Set SMB Protocol to 3.0 to start
Set Share Nameto the share name plus the base path without leading or trailing /
Set Username to the user you created (and be case sensitive)
Set Password to the password you created (and it is also case sensitive)
Set Root Directory to /
Click Test Share
If you get failed to mount, change the SMB protocol as needed
Check that Share Name doesn't have leading ot trailing /

SierraArgo, 2023-03-13:
Awesome, looks good, thanks my friend!

starbasessd, 2023-03-13:
But did it work?

SierraArgo, 2023-03-13:
Sure did!

starbasessd, 2023-03-13:
Don't forget to click Apply at the top after testing works...

yaomtc, 2023-10-20:
I had motioneyeOS installed initially, before I found this issue, but the wiki seems to be for the python2 version, not the dev branch... I figured motioneye_init would have started the web server, but unlike before, putting my Raspberry Pi's IP address in doesn't bring up the motionEye interface yet.

starbasessd, 2023-10-20:
@yaomtc What are you trying to get help for?
motionEyeOS is stand alone with a specific version for specific hardware. Its GUI is at http://<ip_address>
motionEye was Python2 until the current dev version, but it is no where near complete. Its GUI is at http://<ip_address>:8765
You should be able to hook up at least a monitor to see what is going on during the boot sequence.

yaomtc, 2023-10-20:
> What are you trying to get help for?

> **putting my Raspberry Pi's IP address in doesn't bring up the motionEye interface yet.**

I don't have any issue booting. I looked at the journal but couldn't find anything specifically about the server having an issue. I can connect to it over SSH via that IP address (10.0.0.3), but when I try to access 10.0.0.3:8765 via my browser, it's unable to connect.

starbasessd, 2023-10-20:
@yaomtc  are you using motionEyeOS or mpotionEye on another OS? If the latter, what is the base OS?
If you are using motionEye on another OS, what does the command
> sudo systemctl status motioneye

give you? one of mine gives:
>[sudo] password for pi:
● motioneye.service - motionEye Server
     Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor pre>
     Active: active (running) since Fri 2023-10-20 17:43:06 EDT; 12min ago
   Main PID: 482 (meyectl)
      Tasks: 31 (limit: 6865)
     Memory: 451.8M
     CGroup: /system.slice/motioneye.service
             ├─482 /usr/bin/python2 /usr/local/bin/meyectl startserver -c /etc/>
             ├─563 /usr/bin/motion -n -c /etc/motioneye/motion.conf -d 5
             └─581 /usr/bin/python2 /usr/local/bin/meyectl startserver -c /etc/>

yaomtc, 2023-10-20:
It's Raspbian Bullseye. I had tried enabling/starting that service previously but it didn't help. After doing that again: 
```
● motioneye.service - motionEye Server
     Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
     Active: active (running) since Fri 2023-10-20 17:00:01 CDT; 1min 1s ago
   Main PID: 22187 (meyectl)
      Tasks: 5 (limit: 1595)
        CPU: 5.251s
     CGroup: /system.slice/motioneye.service
             ├─22187 /usr/bin/python3 /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
             └─22233 /usr/bin/python3 /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf

Oct 20 17:00:01 rpi systemd[1]: Started motionEye Server.
Oct 20 17:00:05 rpi meyectl[22187]: configure_logging cmd motioneye: False
Oct 20 17:00:05 rpi meyectl[22187]: configure logging to file: None
Oct 20 17:00:05 rpi meyectl[22187]:     INFO: hello! this is motionEye server 0.43.0
Oct 20 17:00:08 rpi meyectl[22187]:     INFO: cleanup started
Oct 20 17:00:08 rpi meyectl[22187]:     INFO: wsswitch started
Oct 20 17:00:08 rpi meyectl[22187]:     INFO: tasks started
Oct 20 17:00:08 rpi meyectl[22187]:     INFO: mjpg customer garbage collector has started
Oct 20 17:00:08 rpi meyectl[22187]:     INFO: server started
```

Still can't load

starbasessd, 2023-10-20:
What does the command
> ss -tulpn | grep 8765

Mine reports:
>  ss -tulpn | grep 8765
tcp   LISTEN 0      128          0.0.0.0:8765      0.0.0.0:*

yaomtc, 2023-10-20:
Same.

    tcp   LISTEN 0      128          0.0.0.0:8765       0.0.0.0:*

starbasessd, 2023-10-20:
Need to open a ticket at https://github.com/motioneye-project/motioneye/issues so the devs can see it.

yaomtc, 2023-10-20:
My mistake, I must have got my tabs mixed up

Ghosti10, 2023-11-09:
**Ok guys , the trick is super easy.**
The Module3 Cam has issues with motioneye.
but it is working perfcet with Spyglass. Here is the trick: Install spyglass, start run.py -r 1920x1080 which is creating a stream. (http://youip:8080/stream) check this out for installation: https://robertmarks.cloud/posts/2023/02/setting-up-pi-camera-module-3-in-mainsailos/ (credits to Rob Marks)

then install Motioneye: https://github.com/motioneye-project/motioneye/wiki/Install-on-Raspbian-Bullseye

then DO NOT ADD MJPEG Cam in motioneye. Add a NETWORK Camera in motioneye,
and you have a perfect working motioneye with motion detection . And YES, autofocus is working as well. Enjoy

bob59, 2024-09-27:
Ghosti10
I have been trying to follow your guide, I have a Raspberry Pi 3B+ running Bullseye
I have installed Spyglass and can see the video stream, then installed Motion and Motioneye, but I am having trouble with adding a camera in Motioneye
for a network camera what would be the URL - same as spyglass maybe ?, also the other required field is camera, but there is nothing in the drop down box
any ideas
thanks
bob