motioneye-project/motioneyeos #2982: No img for Raspberry Pi 0W

lindsayfowler, 2023-03-11:
It was suggested in this [post](https://github.com/motioneye-project/motioneyeos/issues/2967) to use [dev20201026](https://github.com/motioneye-project/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi-dev20201026.img.xz) for Raspberry Pi 0W. 

I used this and it hasn't worked yet. 

Would it be possible to label the img that is compatible with RPi0W somewhere in the file name?
Thanks

starbasessd, 2023-03-11:
What issue are you having? 
Which PiZero do you have? On the bottom, just below the holes for the GPIO you will see a line of white text. It will begin 
"Raspberry Pi Zero" Please include the rest of that line here in your response.
If you are using the Raspberry Pi Imager program, and have used it previously to include your settings, it breaks the motionEyeOS image. You will need to use Etcher or Win32DiskImager instead.
If you connect a monitor to it, does it stop at the 'rainbow screen' or do you get a 'panic rebooting' message?
Do yo include the wpa_supplicant.conf file?

acarrou, 2023-04-08:
The Issue I had was it was getting stuck and trying to panic reboot. I had to make sure the file name was "wpa_supplicant.conf" and had the correct wifi credentials and it worked. Check[ here](https://github.com/motioneye-project/motioneyeos/wiki/Wifi-Preconfiguration) for more info.

alkimient, 2023-06-09:
this didn't work for me.
I included the wpa_supplicant.conf file, but it still does the same thing.
It panic reboots because it can't find the wired network device.

starbasessd, 2023-06-09:
Give the text under the GPIO pins on the back of the Pi.

alkimient, 2023-06-09:
Raspberry Pi Zero W V1.1
I don't think any of the Raspi Zeros have a wired network interface.

starbasessd, 2023-06-09:
No, they do not, unless you have a USB dongle type Ethernet adapter.
Does it work with a RaspberryPiOS and the same wpa_supplicant.conf file? If not, then I would compare the file to a good one, like: 
- - - - -
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
    ssid="SSID"
    psk="P@sSW0Rd"
}
- - - - -
country needs to be YOUR country 2 letter code
ssid needs to be your Access Point name, it is case sensitive, and should NOT have special characters
psk needs to be your password, it is case sensitive, and may have issues with some special characters

alkimient, 2023-06-09:
Now it works.
I had to put the wpa_supplicant.conf file in the root directory, not /boot/
I also converted the file encoding to ANSI, but I'm not sure if that was necessary.

starbasessd, 2023-06-09:
/root once booted is read-only. /boot is the first partition, read/writable by windows (fat32) and partition 2 /root and partition 3 /data are read-only ext4 partitions after first boot.