motioneye-project/motioneyeos #2985: Cannot upload pictures to Google Drive or Photos, cannot obtain Key

derletztename, 2023-03-21:
Hi
I am new to Motion OS and try to set up a simple system to watch birds in a bird house with the kids. Streaming works just fine, but I somehow fail automatically upload the pictures. I tried with Google drive and photos, both times google tells me: "Access blocked: The request from motionEye is invalid"

Fehler 400: invalid_request
The out-of-band (OOB) flow has been blocked in order to keep users secure. Follow the Out-of-Band (OOB) flow migration guide linked in the developer docs below to migrate your app to an alternative method.
Anfragedetails: redirect_uri=urn:ietf:wg:oauth:2.0:oob
[Zugehörige Dokumentation für Entwickler](https://developers.google.com/identity/protocols/oauth2/resources/oob-migration)

Probably that helps you finding the issue, I have no idea what to do :D
It feels like the software needs to be adjusted to things google has changed recently, so it must effect everybody not just me

Many thanks for developing MotionOS
Best
Paul

### Preliminary Docs
I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606.

### Board Model

I am using a Raspberry PI 3B+.

### Camera

I am using the following type of camera: MMAL Camera.

My camera model is: RasPi Cam v2.1.

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi and Ethernet.

### Peripherals

none

starbasessd, 2023-03-21:
Known issue, as is a similar issue with Drop Box. Google and DB both upgraded their security, breaking programs like motionEyeOS. Until/if/when the new devs get around to updating motionEyeOS, this will remain 'broken'. There are ways to connect to Google Drive and Drop Box if you use [motionEye on another OS](https://github.com/motioneye-project/motioneye/wiki/Installation) 
You'll have to do some research on your own to find the links to instructions on how to do it in Linux...

derletztename, 2023-03-22:
Thank you for the answer. I only chose Google because I expected it to be quick and easy. As it currently does not seem to be quick and easy, I pivoted and try to upload via FTP to my local NAS system. But that also failed.
I cannot find documentation that help me get it working. It seems as I managed to connect motionEyeOS to the NAS via FTP, but somehow the filename seems to be off. 
I get the message: "Accessing the upload service failed: 550 192.168.178.34: Forbidden filename!"
I wrote "192.168.178.34/Public/BirdCam" into the Location folder.
I don't understand what exactly is wrong and cannot find proper instructions on that topic.
Could you maybe help me out with that?

derletztename, 2023-03-22:
Thank you very much for your help. It is working now :)
I always had something in front of the /Public/BirdCam and therefore got the error. 

I mean the original issue hasn't been fixed, so technically the issue should not be closed, but as nobody will work on it soon and when someone will rework MotionEyeOS, they will rework the Google upload anyways, we could just close this issue, or what do you think?

starbasessd, 2023-03-22:
You can close it.

derletztename, 2023-03-22:
Problem still existing, but because what I wrote before, I will close the issue anyways.

closed 2023-03-22 by derletztename