motioneye-project/motioneyeos #2986: Cloud service - instead GoogleDrive

myscho15, 2023-04-01:
I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

20200606

### Board Model

Raspberry PI 3B+

### Camera

My camera model is: RpiHQ cam

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi - 4G Router

Hallo all,

first sorry for my bad English.

I see many issues about problem with Google Drive bug and the bug is not fixed. I use MotionEyeOS on everyday to protect and monitor my Cabin and it is very good software, but i am not good with Rpi and programming. But i found here in issues the possible solution with use Rclone.com and idrivee2.com from @xjon685 and the solution from @penguinjeff .

I hope that is here more people like me, they can hard to understand how make solution in thread #2954 to work. 

Can anyone PLEASE make guide to make how make this solution to work or guide (easy understand) to make another cloud based solution that will work on phone? Payment for cloud service is no problem. Also i can pay anyone here to make me guide to make solution to work MotionEyeOs to work with Google drive or another solution. Thank all very much. 

I found this solution but i dont know if will work https://github.com/goldfix/motioneyeos_ext

starbasessd, 2023-04-01:
It's not a bug in motionEyeOS, it was a change that Google and DropBox both made to their security. The new devs haven't gotten around to make a new version of motionEyeOS that uses only Python3 yet.

myscho15, 2023-04-01:
Ok sorry for bug/Google security improvement. 

But the question is i am looking for some help with finding solution how can MotionEyeOs can somewhere upload send photos and i can see this photos. Email solution is not good but better as nothing. 

In MotionEyeOs are possibilities Network share (SMB share) or upload Media files to FTP or SFTP server... but i dont know how make this work... :-(

starbasessd, 2023-04-01:
You need a service (like Google Drive) where you can upload using either of those methods, but I would recommend SFTP over FTP or SMB on the internet, as the SMB and FTP protocols are more oriented to home networks. AWS has SFTP support (or did last time I checked. Most do it yourself web hosts also support it.

myscho15, 2023-04-01:
Yes! That is what i am looking for, but i am looking for anyone, who know explain me, what must i do and how setup. :-)

Jaizan, 2023-12-30:
I have the same problem.  
**I need to be able to see CCTV images remotely**.    Google Drive satisfied that requirement for years, but no longer works.

So I'm trying to figure out what a good and alternative solution would be.

1  From what I read, alternative CCTV Pi based systems are very difficult to set up

2  I'm using MotionEye OS.   So to use something like RCLONE on the same device, presumably I need both MotionEye and RCLONE running on another OS on the PI ?

3  I see MotionEye has the option to save to a NAS.  I think I could set one of those up using another Pi.   Then, presumably I could either fiddle around until I can access that via the internet or put RCLONE on it to upload to the internet ?   

What's the easiest work around ?
Whilst I always appreciated MotionEye, I just didn't realise how much until it has reached end of life !

starbasessd, 2023-12-30:
Both options are available and viable.
1) yes, you can set up another PI to act as an intermediary, either as an SMB NAS to then upload to a cloud service or
2) you can set up another device to act as a VPN or secure port forward and storage.

Unfortunately motionEyeOS doesn't support either option internally, but can upload to SMB, FTP or SCP/SFTP NAS with little to no issues.

I have both of these options set up (using option 2 on as an-needed basis, not 24/7) and which is better for you depends on several factors, including your skill level with Linux, your preferences as to OS (I prefer Debian Lite, myself, but I cannot recommend the more esoteric OSs), and your available hardware. 
If you go the motionEye on another OS route, you need to stick with a somewhat older OS (Debian Buster (10) or Bullseye (11) based) as motionEye is Python2.7 based and none of the current OSs support it anymore. 
My 'primary servers' for motionEye are x86 based, (2 Lenovo M73 TFFs, 16GB/1TB i5s, and VMs on a Ryzen 7 64GB/14TB) but I can spin up motionEye or motionEyeOS on any Pi from a Pi1 through a Pi4-4GB or 8GB along with a few PiZeros. I don't have any Pi02Ws or Pi5s (wouldn't be supportable as it only has RPiOS Bookworm (12)). 

If you need help or ideas, let me know.

Jaizan, 2024-01-07:
What I was actually hoping to do was:
1  Ditch MotionEye OS
2  Install Raspian (or possibly another OS)
3  Install Motion Eye on the same device, running on Raspian
4  Install RCLONE on the same device, also running on Raspian

Is that possible ?   

I thought it wouldn't be too difficult, so decided to try. 
Ububtu seemed slow, so I decided to go with Raspian.
However, I'm getting some unexpected errors at step #3 currently, when I wasn't really expecting any difficulty until step 4.

Suppose I'd better just check that what I want to do will work before investing too much more time on this.

starbasessd, 2024-01-07:
Can / did / does it work: yes.
You need to use the [Buster](https://downloads.raspberrypi.com/raspios_oldstable_lite_armhf/images/raspios_oldstable_lite_armhf-2021-12-02/) (need CSI camera) or [Bullseye](https://pimylifeup.com/download-raspbian/) (at the bottom) (only network or USB cameras) Raspbian images.
The latest Debian 12 (Bookworm) and derivatives (Latest Raspbian, Ubuntu 23.x, Mint, etc) no longer have Python2.7 in the repositories. That is what the error in #3 involves.

Jaizan, 2024-01-07:
Thank you.
I'm already using Bullseye, due to your other recent advice on that.

I decided to start again, with a second fresh install of Bullseye.
I got this error message after pip2 install motioneye

     ERROR: Command errored out with exit status 1:
     command: /usr/bin/python2 -c 'import sys, setuptools, tokenize; sys.argv[0]                                                                                                                                                                                                                                              = '"'"'/tmp/pip-install-_BBy0k/pycurl/setup.py'"'"'; __file__='"'"'/tmp/pip-ins                                                                                                                                                                                                                                             tall-_BBy0k/pycurl/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__fil                                                                                                                                                                                                                                             e__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(                                                                                                                                                                                                                                             code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-Y7H4                                                                                                                                                                                                                                             Tv
         cwd: /tmp/pip-install-_BBy0k/pycurl/
    Complete output (11 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-_BBy0k/pycurl/setup.py", line 961, in <module>
        ext = get_extension(sys.argv, split_extension_source=split_extension_sou                                                                                                                                                                                                                                             rce)
      File "/tmp/pip-install-_BBy0k/pycurl/setup.py", line 623, in get_extension
        ext_config = ExtensionConfiguration(argv)
      File "/tmp/pip-install-_BBy0k/pycurl/setup.py", line 101, in __init__
        self.configure()
      File "/tmp/pip-install-_BBy0k/pycurl/setup.py", line 238, in configure_uni                                                                                                                                                                                                                                             x
        raise ConfigurationError(msg)
    __main__.ConfigurationError: Could not run curl-config: [Errno 2] No such fi                                                                                                                                                                                                                                             le or directory
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check th                                                                                                                                                                                                                                             e logs for full command output.

starbasessd, 2024-01-07:
Checking right now.

starbasessd, 2024-01-07:
Confirmed [install](https://github.com/motioneye-project/motioneye/wiki/Install-On-Raspbian) and operation on Raspbian 10 (Buster) with latest updates.
Currently esting Bullseye

starbasessd, 2024-01-07:
Confirmed [install](https://github.com/motioneye-project/motioneye/wiki/Install-on-Raspbian-Bullseye) and operation on Raspbian 11 (Bullseye) with latest updates.
[image]

starbasessd, 2024-01-07:
If you do Step 3 as a single copy/paste it will have issues. If you didn't shift to superuser (sudo su) you will have issues.

Jaizan, 2024-01-08:
Thank you, again !
I have done step 3 as individual steps
I used SUDO -i as in the instructions, **so that may be the problem**.

I tried SUDO SU and repeated some of the commands.    The errors went away, but Motioneye wasn't installed at the end of it.

I started from scratch with a new Bullseye install.
This time it seemed to run through everything with no obvious errors (file)

Then no sign of motioneye operating or in the GUI menu
I reboot, still nothing
I type motion in the terminal.   Nothing
I type motioneye in the teminal. Nothing   
[Motioneye install terminal 2024-01-08 good.docx](https://github.com/motioneye-project/motioneyeos/files/13858530/Motioneye.install.terminal.2024-01-08.good.docx)

starbasessd, 2024-01-08:
2 things I noticed:
You are using the 64 bit version of RaspbianOS. There are significant differences between ARMHF and ARM64 that were not tested, as, at the time, 64bit was 'experimental' for Raspbian. I will test in a little bit, after the caffeine has kicked in.

Command from your log:
> root@pi4-2GB:/home/jeff# wget https://github.com/Motion-Project/motion/releases/                                                             download/release-4.3.2/buster_motion_4.3.2-1_arm64.deb

Should have been:
> wget https://github.com/Motion-Project/motion/releases/download/release-4.3.2/pi_buster_motion_4.3.2-1_armhf.deb

You grabbed the wrong version of motion, and it could not install it correctly. You would need to grab the pi_buster_motion_4.3.2-1_arm64.deb package but the motion project people did not build a 64 bit package for either buster or bullseye. You could try [4.5.1](https://github.com/Motion-Project/motion/releases/download/release-4.5.1/bullseye_motion_4.5.1-1_amd64.deb) but no guarantees. 4.4.0 had major issues, and was not recommended. 

The known working motionEye is on the 32bit Raspbian.

starbasessd, 2024-01-08:
Another option would be to [build motion from source](https://motion-project.github.io/motion_download.html) in the 64 bit environment for pi, but that is beyond the scope of my free support.

starbasessd, 2024-01-08:
@Jaizan Starting the build on ARM64 Raspbian.

Jaizan, 2024-01-08:
Thank you again !

This time I tried the 32 bit version.
Oddly, when authorising SSH in raspi-config, it said there was no default user, but I put my usual user name in & SSH worked.
I did not get this on the 64 bit versions

The install seemed to go OK.    Quicker than previously.     
However, again no visible sign of MotionEye on the PI desk top, including after a reboot*.   From the instructions, I would expect it to run automatically.  
I type motion in the terminal and from all the messages, in terminal, it looks like it's running, checking for cameras etc.
What I was expecting to see is the MotionEye menu, as on my old MotionEye OS & I do not.   Also no sign of it as an installed programme.

*I'm running the terminal window on my PC with SSH, but when I go to the desktop, that's on my TV monitor directly connected to the Pi via HDMI.

starbasessd, 2024-01-08:
With the understanding that NONE of the 64 bit OSs were tested or supported, and out of my own curiosity, I did the following:

Clean install Bullseye ARM64
sudo apt update && sudo apt upgrade -y   ## completed successfuly
sudo su
apt-get install ffmpeg libmariadb3 libpq5 libmicrohttpd12 -y ## 155 pkgs installed
 wget https://github.com/Motion-Project/motion/releases/download/release-4.3.2/pi_buster_motion_4.3.2-1_armhf.deb   ## grabbed successfully
dpkg -i pi_buster_motion_4.3.2-1_armhf.deb   ## install failed
**_apt install --fix-missing ## lists a number of missing dependencies
apt install --fix-broken  ## installs the missing dependencies
dpkg -i pi_buster_motion_4.3.2-1_armhf.deb   ## install succeeded_**
systemctl stop motion  ## succeeded
systemctl disable motion   ## succeeded
apt-get install python2 python-dev-is-python2 -y   ## succeeded
curl https://bootstrap.pypa.io/pip/2.7/get-pip.py --output get-pip.py   ## succeeded
python2 get-pip.py   ## succeeded
apt-get install libssl-dev libcurl4-openssl-dev libjpeg-dev zlib1g-dev -y   ## succeeded
pip2 install motioneye   ## succeeded
mkdir -p /etc/motioneye   ## succeeded
cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf   ## succeeded
mkdir -p /var/lib/motioneye  ## succeeded
cp /usr/local/share/motioneye/extra/motioneye.systemd-unit-local /etc/systemd/system/motioneye.service  ## succeeded
systemctl daemon-reload  ## succeeded
systemctl enable motioneye  ## succeeded
systemctl start motioneye  ## succeeded
The 3 **_BOLD ITALICIZED_** lines allow it to work in the 64 bit environment. Install the 32bit version of PI Motion, NOT the 64 bit ARM version. The 64 bit ARM version is for generic ARM based PCs, the PI_motion version is specific for the PIs, like the motionEyeOS is compiled for specific boards.

starbasessd, 2024-01-08:
Another difference in motionEye vs motionEyeOS is which port to use.
motionEyeOS http://ip_address
motionEye http://ip_address:8765

starbasessd, 2024-01-08:
> The install seemed to go OK. Quicker than previously. However, again no visible sign of MotionEye on the PI desk top, including after a reboot*. From the instructions, I would expect it to run [...]

motionEye is browser based, not desktop based. You can open the default web browser and go to http://localhost:8765 and it should come up.

In ssh or a terminal window run the following to confirm motionEye is running:
>root@raspberrypi:/home/pi# sudo systemctl status motioneye
● motioneye.service - motionEye Server
     Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2024-01-08 14:34:42 GMT; 16min ago
   Main PID: 13406 (meyectl)
      Tasks: 5 (limit: 779)
        CPU: 4.924s
     CGroup: /system.slice/motioneye.service
             ├─13406 /usr/bin/python2 /usr/local/bin/meyectl startserver -c /etc/motioneye/motioney>
             └─13416 /usr/bin/python2 /usr/local/bin/meyectl startserver -c /etc/motioneye/motioney>

The Active line should be green
you can also run:
>root@raspberrypi:/home/pi# sudo ss -tulpn | grep 8765
tcp   LISTEN 0      128          0.0.0.0:8765       0.0.0.0:*    users:(("meyectl",pid=13406,fd=12))

Jaizan, 2024-01-08:
Thank you very much!    That's working.     
Also, I can view the photos from the one camera that's installed (so far) on the Pi.

So theoretically, I could now install something like RCLONE on the same Pi to upload files from this directory to google drive ?

starbasessd, 2024-01-08:
Theoretically, yes. Outside the scope of my support, though.