motioneye-project/motioneyeos #2987: RPi Zero W hangs on *Starting motion eye:

ThunderClap86, 2023-04-07:


MichaIng, 2023-04-07:
It is probably not the issue, but be careful when changing something via RPi Imager on any other image than official RPi OS. Probably better to use a flasher which has not even the capability to touch anything within the image file, like Rufus or balenaEtcher.

ThunderClap86, 2023-04-07:
Okay, Ill try to give the image file to Rufus and see if that corrects my issue.

ThunderClap86, 2023-04-08:
> It is probably not the issue, but be careful when changing something via RPi Imager on any other image than official RPi OS. Probably better to use a flasher which has not even the capability to touch anything within the image file, like Rufus or balenaEtcher.

[image]

Formatted and re-imaged using unmodified file through Rufus. 

Needed wpa_supplicant.conf to get past the first hangup with wired configuration. It made it to the same hang. Not sure what to try next.

starbasessd, 2023-04-08:
I recommend [dev20201026 for RPiZero](https://github.com/motioneye-project/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi-dev20201026.img.xz)

ThunderClap86, 2023-04-08:
> I recommend [dev20201026 for RPiZero](https://github.com/motioneye-project/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi-dev20201026.img.xz)

I am re-imaging with rufus now. Ill update in a few when it's done and powered on.

ThunderClap86, 2023-04-08:
> I recommend [dev20201026 for RPiZero](https://github.com/motioneye-project/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi-dev20201026.img.xz)

This image seems to have fixed the issue. Thank you!

closed 2023-04-08 by ThunderClap86