motioneye-project/motioneyeos #2991: Can not get my camera to work with motioneyeos on my pi 4

BradysKool, 2023-05-04:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606

### Board Model

I am using the following board/model Raspberry PI 4B.

### Camera

 I am using a raspberry pi model 2 camera that I bought off of amazon.

here the link to it: https://www.amazon.com/dp/B01ER2SKFS?psc=1&ref=ppx_yo2ov_dt_b_product_details 

### Network Connection

My motionEyeOS unit is connected to the network via:  Ethernet.

### What I have done

I installed motionEyeOS onto my raspberry pi. When I booted it up it looked normal, but it has a few problems, one is that when i type in the IP address into the browser the page comes up but I cannot add the camera so that I can view it. Since the camera is plugged into directly into the board I thought I would try to configure the MMAL camera but whenever I click on add MMAL camera it says "An error occurred. Refreshing is recommended". Because of this I looked up things on how to see if your camera is supported and it suggest typing in a command to see if the camera is connected and supported. But my keyboard does not work. This is weird because it works fine if I am using Raspbian OS. I then went and bought a USB camera to see if it works and I and no luck on getting it to work.

I am new to motioneyeos so I might be doing something wrong. Can I please have some guidance on what to do?

starbasessd, 2023-05-04:
Use image [dev20201026](https://github.com/motioneye-project/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi4-dev20201026.img.xz)
Make sure the camera is working and is available:
> vcgencmd get_camera

It should show 
>  supported=1 detected=1

If you do not get that response, then you probably have the ribbon cable backwards, or the camera is defective.

BradysKool, 2023-05-04:
Thank you for the response,

how do I run the command; my keyboard does not work on my pi while I am in motioneyeos is there a different thing I can do?

starbasessd, 2023-05-04:
If you have a keyboard & monitor attached, you sign in with 'admin' and whatever password you set. If not, you can use SSH and remote into the console with something like PuTTY from Win10 or 11, or ssh from MacOS or Linux...

BradysKool, 2023-05-04:
Ok thank you I will try to use SSH. Thank you for the suggestion I will get back to you later on how it works!

BradysKool, 2023-05-05:
Ok so I used SSH to get into the pi and I ran the vcgencmd get_camera command and it gave the response:

supported=0 detected=0

I got the same result for my usb camera as I did my pi cam. I tried putting the ribbon cable in the other way but I at the same response every time.

This is weird because the camera works perfectly when I am in Raspian OS. Also, my system will not pick up my keyboard or mouse. Do you think this has something to do with the version of motioneyeos?

BradysKool, 2023-05-05:
Sorry for the confusion I got it all fixed. I was installing the wrong version. Thank you though for all the support.

closed 2023-05-05 by BradysKool