motioneye-project/motioneyeos #2993: Motioneye is not detecting Logitech C270 webcam.

beug99, 2023-05-21:


starbasessd, 2023-05-21:
@MichaIng @beug99 
> I'll move this to the motionEyeOS repository. To troubleshoot: Does this command exist on motionEyeOS, and if so, what does it return?
> 
> ```shell
> v4l2-ctl --list-devices
> ```

 Yes it does, but did you ask what kind of camera he is attempting to add? Network vs USB? (400 doesn't have a CSI interface)

MichaIng, 2023-05-21:
"Local V4L2 camera" sounds like a USB camera.

Elfdawn, 2023-05-27:
I'm having the exact same issue on a Raspberry Pi 400 with another USB webcam.  I was having issues with getting Motioneye to boot so I followed @starbasessd 's advice from nearly 3 years ago. https://github.com/motioneye-project/motioneyeos/issues/2419#issuecomment-646568538
This worked, but as warned futher on, USB cameras seem to stop working with this fix. I have noticed that when i look at the camera settings, the camera channel device is specified as dev/video10. From testing the Camera earlier on rasbian, i used dev/video0 to view the camera. Is there a way i can change the "Camera Device" setting to this channel?

closed 2023-05-28 by beug99