motioneye-project/motioneyeos #2995: My camera isn’t working

OtherLinux, 2023-06-03:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20201026.

### Board Model

I am using the following board/model: 4B 1 gb.

### Camera

I am using the following type of camera: MMAL.

My camera model is: Pi cam v2.

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi.

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

none

So good day to you all I’m having trouble with setting up my camera on motioneye. Whenever I try it just gives an unknown error and recommends to refresh

starbasessd, 2023-06-03:
A CSI (ribbon cable connected) camera should be auto detected. If you connect via ssh or use the monitor/keyboard console, and run the command:
>[root@meye-4cec78b8 ~]# vcgencmd get_camera

You should get:
>supported=1 detected=1

If not, then there could be several issues:
bad ribbon cable
ribbon cable not connected correctly (wrong direction at one end or the other)
bad camera

You can double check by installing RaspberryPiOS Buster (Legacy) and running that command there, or running any number of camera apps to see if it works there.

OtherLinux, 2023-06-03:
I tried the command multiple times but it always said supported=0 detected=0, libcamera interfaces=0. When I used the command on rpios the newest one it worked and so did libcam. I read almost every camera related issue on this forum but I could not find a way to solve it

starbasessd, 2023-06-03:
Another set of test commands to run:
> cd /data/output
> service motioneye stop
> raspistill -o output.jpg

It should come back to the prompt without an error, and if you list the directory, you should see the file 'output.jpg'
you can restart the services with:
> service motioneye start

OtherLinux, 2023-06-03:
So this is the response from the raspi
mmal: Cannot read camera info, keeping the defaults for OV5647
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
mmal: Failed to create camera component
mmal: main: Failed to create camera component
mmal: Camera is not enabled in this build. Try running "sudo raspi-config" and ensure that "camera" has been enabled

starbasessd, 2023-06-03:
libcam is the Bullseye replacement. motioneyeOS is written with buster and python 2.7. Need to test with the Legacy image.

OtherLinux, 2023-06-03:
I tried it with raspistill
didn't work

starbasessd, 2023-06-03:
> So this is the response from the raspi mmal: Cannot read camera info, keeping the defaults for OV5647 mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM) mmal: [...]

Where do you get this error? In motionEyeOS dev20201026 or RasberryPiOS? Which version of RPiOS if using that?

OtherLinux, 2023-06-03:
The motioneyeos dev20201026

starbasessd, 2023-06-03:
Try re-imaging the SDCard, or use a different SDCard. You shouldn't be getting the error:
> mmal: Camera is not enabled in this build. Try running "sudo raspi-config" and ensure that "camera" has been enabled

as that function is already enabled and raspi-config isn't installed in the image.

OtherLinux, 2023-06-03:
I tried re-imaging the SDCard atleast 5 times to make it work and not once did it work

starbasessd, 2023-06-03:
Have you tried the 
https://downloads.raspberrypi.org/raspios_oldstable_armhf/images/raspios_oldstable_armhf-2023-05-03/2023-05-03-raspios-buster-armhf.img.xz
image? 
(enable camera in raspi-config, then test with the raspistill -o output.jpg command)

OtherLinux, 2023-06-03:
alright i'll try that

OtherLinux, 2023-06-03:
[image]
kinda not working

starbasessd, 2023-06-03:
I just downloaded it, getting ready to image a SDCard.

OtherLinux, 2023-06-03:
I'm downloading it so I will report as soon as I can

OtherLinux, 2023-06-03:
Wait raspberrypi has 2 TLD's .org and .com?

starbasessd, 2023-06-03:
Yes. .org is the 'educational' wing, and .com is the software (OS) and hardware side...

starbasessd, 2023-06-03:
They split it like 2 years ago...

OtherLinux, 2023-06-03:
Oh i didn't know that

OtherLinux, 2023-06-03:
Everything worked in the raspios
The camera everything

starbasessd, 2023-06-03:
re-download motioneyeOS from here:
https://github.com/motioneye-project/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi4-dev20201026.img.xz
and try again. _**Be sure NOT to use raspi-imager**_, use Etcher or Win32DiskImager...

OtherLinux, 2023-06-03:
[image]
Still the same

OtherLinux, 2023-06-03:
Still the same
[image]

OtherLinux, 2023-06-03:
> re-download motioneyeOS from here: https://github.com/motioneye-project/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi4-dev20201026.img.xz and try again. _**Be sure NOT to use raspi-imager**_, use Etcher or Win32DiskImager...

wait i use the raspi imager sorry

starbasessd, 2023-06-03:
You can't 'add' it, it should be auto-found and auto set up at startup.
raspi-imager breaks the SDCard, especially if you have your wifi pre-set in it. 
If you don't have ethernet at least for the first boot, you need to copy the wpa_supplicant.conf file to the /boot (first) partition before forst boot.

OtherLinux, 2023-06-03:
So I used the image burner that comes with my linux distro and it still doesn't work

starbasessd, 2023-06-03:
Your setup may be one of the few that I can't figure out. I have a v2 PiCam on a Pi4 4GB and 8GB, that is working fine. Works on a PiZeroW, Pi3B+ and  others, too, all using dev20201026. I don't know. Might be a modified/updated camera that wasn't supported in the kernel from 2020.

OtherLinux, 2023-06-03:
And what did you do differently than I?

OtherLinux, 2023-06-03:
But still thanks a lot for the help

starbasessd, 2023-06-03:
I don't know.
I downloaded the dev20201026 image
I copied it to a 16GB SDCard using Win32DiskImager
I copied wpa_supplicant.conf to the first partition (/boot)
I ejected it and inserted into a Pi4 4GB with an attached PiCam V2
I booted, watched it set itself up and reboot
I went to the WebGUI interfaca as 'admin' and the camer was showing.
I tested with the PiZero image and PiCam V2, ssame process, same result.
Used dd command to image another SDCard with the same image for the Pi4 in Debian Linux (Bullseye)
same process, same result.

OtherLinux, 2023-06-03:
Wait I’m not that sure of one thing could the problem
be with my limited ram

starbasessd, 2023-06-03:
Which Pi4 do you have? 1GB, 2GB, 4GB, or 8GB? Considering it runs on a PiZero (which only has 512 MB, and the Pi1s with also 512MB, RAM is not usually a limitation.

OtherLinux, 2023-06-03:
Well I have the one gig model

OtherLinux, 2023-06-03:
Im like unsure why it isnt registering the camera

starbasessd, 2023-06-03:
Should not be an issue.

OtherLinux, 2023-06-03:
Well I’m kind of like wth why isnt it working

starbasessd, 2023-06-03:
Might be a modified/updated camera that wasn't supported in the kernel from 2020, if it not an Official RPi Camera V2 (one of the other manufacturers/clones)

OtherLinux, 2023-06-03:
Well it came from an official box and It was supported in the old raspberry os

starbasessd, 2023-06-03:
I don't know. Leave it up, maybe someone else has suggestions...

OtherLinux, 2023-06-03:
Thank you for the responses

starbasessd, 2023-06-03:
Here's a link to a much older raspberrypios image with a much older kerenel: 
https://downloads.raspberrypi.org/raspios_oldstable_lite_armhf/images/raspios_oldstable_lite_armhf-2021-12-02/
If you want, for grins and giggles, try it (enable the camera in raspi-config, and run the 
> vcgencmd get_camera
and
> raspistill -o output.jpg

and see if it responds correctly. the line in your error messages still concerns me: 
"mmal: Cannot read camera info, keeping the defaults for OV5647"

OtherLinux, 2023-06-03:
OV5647 that sounds like it could be the problem but i dont know

OtherLinux, 2023-06-03:
> So this is the response from the raspi mmal: Cannot read camera info, keeping the defaults for OV5647 mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM) mmal: [...]

Why the hell did it say "mmal: **### Camera is not enabled in this build**. Try running "sudo raspi-config" and ensure that "camera" has been enabled"

OtherLinux, 2023-06-03:
Im so confused right now

starbasessd, 2023-06-03:
Either it's newer hardware, or newer firmware in the camera, or it's one of the HQ cameras, and simply not recognized by the kernel in the motionEyeOS image. The error message is one that was built into the kernel. I've also seen that error from some of the chinese clone cameras of v1.0 and v1.3. Around the time of v2.0, there were the v2 HQ cameras (really high megapixel counts) that came out just after dev20201026, and those were not supported. There hasn't been an update since. 
If you want a possible work around, you can try going with motionEye on RaspberryPiOS Buster Legacy following the [instructions here](https://github.com/motioneye-project/motioneye/wiki/Install-On-Raspbian) The RPiOS Bullseye still doesn't support CSI but does work with USB and Network cameras.

OtherLinux, 2023-06-03:
Thanks for the answer I have however I'm thinking of buying an USB camera instead