motioneye-project/motioneyeos #2996: Motioneye not booting, Bios says: Firmware not compatible

FootsoreBird03, 2023-06-11:
Problem: Pi won't boot with the OS.

Details:
I Downloaded the image and unziped using 7-zip. I then installed the image using raspberry Pi imager v1.7.5. When plug in power, after the SD card is already in, and the pi does not boot. It instead display's the error (picture 1). Please note this is my first rasberry pi project so any help given will need the "extra steps" so I can fiqure out what the help even means. Any youtube videos on this problem would be estatic. Please note, I saw a post about this from 2022, but I have no idea how to implement the given advice to fix the issue. 
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606

### Board Model

I am using the following board/model: Raspberry pi 4B 4gb

### Camera

I am using the following type of camera: None, trying initial setup with waiting for the camera to arrive

### Network Connection

My motionEyeOS unit is connected to the network via: None, trying to access it directly via an HDMI connection)

Images: 

This is what the boot screen displays 
[image]

Then goes to this for a sec, than back to the boot error
[image]

My hardware as attatched
[image]

Taxicletter, 2023-06-25:
I tried headless with a Raspberry Pi 3 & 4, a few times (connected with ethernet-cable). After a blinking green light (maybe 3 times) I couldn't see any activity and I got no IP-address...
I think the image doesn't work anymore.

starbasessd, 2023-06-25:
It's a known issue, and depends on which version of the bootloader is installed in the Pr4. It does not affect The Pi3 and Pi4. versions are different. You need to use the correct image for the device. There are several threads on which bootloaders are tested with which versions of the Pi4 (2GB, vs 4GB, vs 8GB) you are using.

Taxicletter, 2023-06-26:
Where exactly can we find those versions? I only know of the page with different images for different boards, not for different Raspberry Pi's.

starbasessd, 2023-06-26:
https://github.com/ccrisan/motioneyeos/releases

starbasessd, 2023-06-26:
For all the boards, I recommend dev20201026 version

Taxicletter, 2023-06-27:
AH! I didn't see "Show all 16 assets" :-)
I also hadn't seen the assets voor 20201026 that way...

Thank you!

Taxicletter, 2023-06-29:
I can confirm the version you suggested does work! (On Rpi3)

Joenazzzz, 2023-08-21:
on my Raspberry Pi 4 8gb it doesnt work...
Ive updated with sudo
and used the newest version but it shows the same issue as @FootsoreBird03

Joenazzzz, 2023-08-21:
any ideas?

Serpentstar, 2023-08-23:
The current repository hasn't been updated since April of last year.  Is this project still undergoing development?

jumboswart, 2023-10-29:
I have the same issue and I am using the Raspberry Pi 4B 4gb in a pironman case, can anyone assist

starbasessd, 2023-10-29:
What is the result of 
> vcgencmd bootloader_version

when running a raspberrypios like buster lite or legacy lite?

sts-ryan-holton, 2024-12-17:
Trying to install on Raspberry Pi 5, which one should I install? The latest one for 4 gives an  OS error

bbillp, 2024-12-18:
Your experienced has warned the rest of us.   I installed ‘motion’ from the pinwheel site mentioned in the installation instructions Onto a Pi-3 but it was not a smooth installation as it was very hard finding the motioneye installation file .  After installation I did a apt-get upgrade/update which erased my installation to replace it with a newest version but it was a download only and it did not install until I found the installation .py file,  all very weird.   Prior to this I used motioneye on an old Pi-2 for years without trouble.    Bill P Sent from my iPad