motioneye-project/motioneyeos #2999: Time camera - auto email notification with photo

myscho15, 2023-06-18:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 2000606

### Board Model

I am using the following board/model: Raspberry Pi 3A+

### CODE:

ssh into the pi with motionEyeOS
run the following commands
cd /data/usr
nano ./systemup.py
copy-paste the following:

import smtplib, ssl
port = 465 # For SSL
smtpServer = "smtp.gmail.com"
sender_email = "[from_email@gmail.com](mailto:from_email@gmail.com)" # Enter your gmail account address
receiver_email = "[to_email@gmail.com](mailto:to_email@gmail.com)" # Enter receiver address (can be the same)
password = "your_gmail_password"
message = "motionEyeOS System is up."
context = ssl.create_default_context()
mailServer = smtplib.SMTP_SSL(smtpServer, port )
mailServer.login(sender_email, password)
mailServer.sendmail(sender_email, receiver_email, message)
mailServer.quit()

0 10,20 * * * python /data/usr/systemup.py

Hallo all, 

in the past [starbasessd] helped me to make auto email notification - works well. #2853 (code above)

But now i try to make some improvement of this new "feature":

- i am normally use motioneyeos on motion trigger setting and photos send via email
- in the specific time (code above) motioneyeos send me email - "I am alive" - this is how it works
 
NOW: i want to make the code that will make in this specific time photo and then send photo via email... like "timer camera".

Is this possible? Can please anyone help me with this? 

Thank all for help.

lil-poppy, 2023-07-09:
The project is pretty old and not actively maintained anymore. What i do is use the things that still work. My process is with motion detection, but it's also possible with interval snapshots. So when motion is triggered i have it se up with a webhook that starts looking for new files on my server. After the motion stops, i have motioneye upload the new files via FTP to my server. Than the script that received the webhook sees the new files (first the image and then the video) and sends me messages with them on telegram. I could just never get email to work and just created a telegram bot for this.

For interval snapshots it would be a bit different, since a webhook is only called on motion detection. If you have another computer i would recommend running a script there and transfer with FTP. If not you can also run it locally. And just check every whatever time you set the interval to for new files and send them via telegram. Or you could make the script to send emails, if that's what you want, i am sure there's a python library for it.

myscho15, 2024-07-08:
> Or you could make the script to send emails, if that's what you want, i am sure there's a python library for it.

Hi, thank you for the answer, last year I was busy I have two little babys. 

So now I have little bit time for play with Rpi. So that’s is I want, so for example in specific time system send me email but with actual photo from camera. Can you please help me with this? I am noob, but I can pay you for help, because I use MotioneyeOS for years, but sometimes happend some malfunction and camera don’t work… (because of cheap CSI cable and extension and weather) because of that I want know if the system works well. I use the Rpi as the online security system on my Cabin. Thank you