motioneye-project/motioneyeos #3000: MotionEyeOs not booting

grandeivid, 2023-07-04:
I am running motionEyeOS version:  motioneyeos-raspberrypi4-20200606.img.xz 

I am using the following board/model:  Raspberry PI 4B - 4 Gb

I am using the following type of camera: Raspberry Camera Module

My camera model is: Original Raspberry Pi Camera Module 3

My motionEyeOS unit is connected to the network via: Ethernet

I am using the following peripherals that I consider relevant to this issue: No peripherals connected

Hi, I downloaded the "motioneyeos-raspberrypi4-20200606.img.xz" image and created a SD card with Etcher. Then, I inserted the card into the Raspberry, plugged the network cable and switched on the Raspberry board. I use "Angry IP Scanner" to find he Raspberry in the network, but it doesn't show up. I can see the led on but I cannot find it in the network.

Sorry, but I am not an expert in this area (I am a Windows engineer). I cannot see what is going on when the Raspberry is starting as I don't have a Micro-HDMI cable.

If anyone can give me a hand with this, I will really appreciate any help.

Thank you in advance,

David

starbasessd, 2023-07-04:
There are several known issues:
1) Version of bootloader. If it is newer than Apr 29 2021 there are known issues. see #2923 and others.
2)  You need to use image dev20201026
3) You might want to install RaspberryPiOS (Buster) (Legacy), install the ssh file in the boot partition, let it fully boot, and try to find the IP Address. Then shutdown, switch to the MotionEyeOS sdcard, wait for 2 minutes, and try  to connect via the same IP_ address.
The hdmi cable will be your best friend when dealing with connectivity issues on a Pi4 as there is no other way to be sure to see what is going on.

grandeivid, 2023-07-04:
Thank you very much,

I don't know what version of bootloader I am using. First, I will try to use the dev20201026 image. If it doesn't work, then I will  find out what version of bootloader I have and try the solution in your point 3.

I'll post the results...

Thanks again

David

Nick268, 2023-07-08:
I have used motioneyeos-raspberrypi4-dev20201026.img.xz to create a card for my 4GB Pi 4. (in fact, numerous different cards were used).

It doesn't boot, I get the messages "start.elf is not compatible" and "this board requires newer software". I'm not sure which bootloader the board has, but the message also says "Firmware Jul 17 2020 11:05:44".

I have read the thread referred to in the post above. (https://github.com/motioneye-project/motioneyeos/issues/2923).
That seems to refer to motion eye installed as an application rather than as the operating system. 

Anyway, as commented in the other thread, should I give up trying to use the motioneyeOS operating system on my Pi4 and stick with an earlier board?

Nick P.

In case I had misunderstood the instructions, I also tried using motioneyeos-raspberrypi-dev20201026.img.xz on the off chance. It didn't work either.

grandeivid, 2023-07-20:
Hi again,

I was trying all possible fixes that I found on the Internet and different forums, but none of them worked out.
So, I decided to install the official Legacy OS from Raspberry and install MotionEye (not the MotionEyeOS). I now I have MotionEye running on my Raspberry Pi 4 !

But... the camera preview doesn't show any picture, instead, I get the error "Unable to open video device". I tried removing the camera and adding the camera again using different codecs, but it didn't work.
I have the Pi Camera Module 3 connected with CSI.

Any ideas on how to have this working will be appreciated.

Thanks in advance,

David

starbasessd, 2023-07-20:
What does:
>vcgencmd get_camera

show?

starbasessd, 2023-07-20:
And did you enable the camera in raspi-config?

grandeivid, 2023-07-20:
Yes, I did enable the camera, actually more than once, just in case.
I will try this command "vcgencmd get_camera" when I get home and post the result.
Thanks

grandeivid, 2023-07-20:
Hi,

This is what I get:

supported=1 detected=1, libcamera interfaces=0

starbasessd, 2023-07-20:
@grandeivid 
Are you sure you started with RPiOS Buster Legacy? 
[It's linked here lite](https://downloads.raspberrypi.org/raspios_oldstable_lite_armhf/images/raspios_oldstable_lite_armhf-2023-05-03/2023-05-03-raspios-buster-armhf-lite.img.xz)
 [It's linked here desktop](https://downloads.raspberrypi.org/raspios_oldstable_lite_armhf/images/raspios_oldstable_lite_armhf-2023-05-03/2023-05-03-raspios-buster-armhf.img.xz)
Your response indicates it is looking for libcamera which is in Bullseye.

grandeivid, 2023-07-20:
Hi, yes.
And I just read that it is not compatible with the Legacy version. So, I quickly formatted the SD card and installed Bullseye 32-bit.

Before trying to install MotionEye, I wanted to check if the camera is working. So, first I enabled the camera in the raspi-config menu and reboot.

Then I type #raspistill -o testshot.jpg# and this is what I get:

mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
mmal: camera component couldn't be enabled
mmal: main: Failed to create camera component
mmal: Failed to run camera app. Please check for firmware updates

With the command #vcgencmd get_camera# I get:

supported=1 detected=1, libcamera interfaces=0

And with this #libcamera-hello#:

error: XDG_RUNTIME_DIR not set in the environment.
Made X/EGL preview window
[0:05:19.037674441] [2135] INFO Camera camera_manager.cpp:297 libcamera v0.0.5+82-2783c8d8
ERROR: *** no cameras available ***

I tried to install the libcamera library with this:

git clone https://git.libcamera.org/libcamera/libcamera.git
cd libcamera
meson setup build
ninja -C build install

But the third and fourth steps gave me the error #command not found#

I guess I have to fix this before installing MotionEye... Do you know how to fix this issue?

Thank you

David

starbasessd, 2023-07-20:
Hmmm... Personally, I wish RPiFoundation would stop breaking things. I found [this](https://forums.raspberrypi.com/viewtopic.php?t=345597) but it looks like until the new devs of motionEye decide they are going to put some effort into moving off of Python2.7 and work libcamera in, it's a lost cause except for older hardware like PiCam 1.x or shifting over to USB & network cameras only.... I wish I could update the documentation, but...

grandeivid, 2023-07-20:
But isn't the camera module 3 compatible with the raspberry 4?
Because I still didn't installed MotionEye. I am just trying to get the camera working directly on the Terminal from the raspberry SO.

starbasessd, 2023-07-20:
You need:
RPi4 2GB, 4GB, or 8GB
Raspberry Pi OS Bullseye 
install RPiOS Bullseye Desktop on SDCard
Boot to the SDCard
run:
>sudo apt update && sudo apt upgrade -y
>sudo raspi-config

enable the camera
save & reboot
Run the following test command:
>libcamera-hello

This should open a preview window and show what the camera is seeing
If it doesn't work, or gives an error, run the 
>vcgencmd get_camera 

command, again, and see if it is working. If not, contact the people you bought the camera from, it may be defective.

starbasessd, 2023-07-20:
@grandeivid 
More info if you need to modify config.txt and [what the basic commands do](https://www.raspberrypi.com/documentation/computers/camera_software.html)

starbasessd, 2023-07-20:
@grandeivid 
More info if you need to modify config.txt and [what the basic commands do](https://www.raspberrypi.com/documentation/computers/camera_software.html)
It would seem that there are several versions of the PiCam3, and some need you to add a line in /config.txt to enable the correct version:
[config.txt mods](https://www.raspberrypi.com/documentation/computers/camera_software.html#if-you-do-need-to-alter-the-configuration)

grandeivid, 2023-07-20:
Thanks again

I followed the instructions on how to check the config.txt and I changed the value of dtoverlay

Originally was: dtoverlay=vc4-fkms-v3d
And I changed to: dtoverlay=imx708

No luck, it did not work, but it did something, because now I got a different error:

libEGL warning: DRI2: failed to authenticate
Made X/EGL preview window
[0:11:18.389751664] [2215]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+82-2783c8d8
[0:11:18.547885981] [2231]  INFO RPI vc4.cpp:437 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media1 and ISP device /dev/media0
[0:11:18.548044462] [2231]  INFO RPI pipeline_base.cpp:1101 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[0:11:18.549981357] [2215]  INFO Camera camera.cpp:1033 configuring streams: (0) 2304x1296-YUV420
[0:11:18.552190751] [2231]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected unicam format: 2304x1296-pBAA
terminate called after throwing an instance of 'std::runtime_error'
  what():  failed to import fd 21
Aborted

Do you have any other idea?

Thank you

starbasessd, 2023-07-20:
You add a second dtoverlay line, not replace it.
Other than that, I have no other ideas...

grandeivid, 2023-07-20:
Wow It worked! I cant believe it :)
Thank you very much

Although, if I try to take a picture with this: raspistill -o testshot.jpg

I still get an error:
ERROR: the system should be configured for the legacy camera stack

But, do you think that I can install MotionEye now?

starbasessd, 2023-07-20:
@grandeivid
raspistill is for Buster. 
Bullseye uses the libcamera commands. 
raspistill and raspivid are not supported by Bullseye. You will not be able to use PiCam v3 with motionEye in Bullseye for this reason. The Python2.7 uses the raspivid and raspistill commands.
Further testing just to test whether the camera is working is to use the libcamera commands:
>libcamera-jpeg -o test.jpg
>libcamera-jpeg -o test.jpg -t 2000 --width 640 --height 480
>libcamera-still -o test.jpg
>libcamera-vid -t 10000 -o test.h264

grandeivid, 2023-07-20:
All the libcamera commands work !

Ok, if MotionEye is not an option then, do you know any other similar software compatible with the Raspberry 4 and the Camera Module 3?
If not, is there a way that I can just connect remotely from my laptop to the Raspberry and start-stop the camera?

Sorry to ask so many questions and thanks again

starbasessd, 2023-07-20:
@grandeivid I am not sure what would fit your needs, as you haven't described them.
As to the commands needed to start / stop, I would still look at motion the program motionEye and motionEyeOS are wrappers for) to control your recording. If I wasn't working on some other projects, I would help you further. 
[Here is a link to the commands for motion](https://motion-project.github.io/motion_config.html)

starbasessd, 2023-07-20:
motion has a web interface of its own to turn on and off features and functions...

grandeivid, 2023-07-21:
Good morning,

"If I wasn't working on some other projects, I would help you further" - You already helped me a lot !

I just want a camera to record matches of padel. The idea is to have a camera pointing to the court and be able to start and stop the recording remotely.

I just took a look to the Motion software that you recommended and it looks that it can do what I need. So, this weekend I will try to install it and set it up.

I will post the results.

Thank you !

aslaxmi, 2023-11-09:
> I have used motioneyeos-raspberrypi4-dev20201026.img.xz to create a card for my 4GB Pi 4. (in fact, numerous different cards were used).
>
> It doesn't boot, I get the messages "start.elf is not [...]

I too have the same issue and now trying to use the application instead of OS