motioneye-project/motioneyeos #3002: Not booting

averagenokiauser, 2023-07-08:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606

### Board Model

I am using the following board/model: Raspberry Pi 4b 4gb ram

### Camera

I am using the following type of camera: 

My camera model is: Trust Spotlight Pro

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

### Log Files

I consider the following log files relevant to this issue: Nothing

Hi, i wrote the image to sd card and edited the image after it was on sd card so it can connect to my Wi-Fi network but after i inserted the sd card into my Raspberry Pi 4b 4gb it doesn't want to boot of the image. It says This board requires newer software. I have also tried writing the image with Raspberry Pi Imager but it didn't help. I would be really happy if someone could help me.

averagenokiauser, 2023-07-09:
Okay now i copied fixup4.dat and start4.elf from latest raspberry pi os and pasted them to motioneye os sd card and then it finally booted up so if you have the same problem just do this

averagenokiauser, 2023-07-10:
Okay now it boots and everything works but it does not detect my usb webcam. I will try plugging in other webcams if i find any

starbasessd, 2023-07-10:
Loading start4.elf and fixup4.dat breaks many of the drivers that USB needs, that is a known issue, and is in several threads. There is no 'easy fix'.

starbasessd, 2023-07-10:
The only bootloader I have tested and confirm works is 29 April 2021. If yours is newer, it is problematic. If it is older there are specific instructions on how to update to 4/29/2021 at Raspberrypi.com and other sites. DO NOT update to the latest and greatest, though.

averagenokiauser, 2023-07-13:
> The only bootloader I have tested and confirm works is 29 April 2021. If yours is newer, it is problematic. If it is older there are specific instructions on how to update to 4/29/2021 at Raspberrypi.com and other sites. DO NOT update to the latest and greatest, though.

I have the latest bootloader on my raspberry pi. Is downgrading a good option and if yes how can i do it?

starbasessd, 2023-07-13:
No, it is not a good option. At this point, I'd recommend RPiOS Buster Legacy if you need to use a CSI camera, or RPiOS Bullseye if you only need Network or USB cameras. Instructions here:
[Buster Legacy](https://github.com/motioneye-project/motioneye/wiki/Install-On-Raspbian)
[Bullseye](https://github.com/motioneye-project/motioneye/wiki/Install-on-Raspbian-Bullseye)
Both have detailed step by step instructions, and I have tested both sets of instructions in the past couple of weeks.

averagenokiauser, 2023-07-13:
> No, it is not a good option. At this point, I'd recommend RPiOS Buster Legacy if you need to use a CSI camera, or RPiOS Bullseye if you only need Network or USB cameras. Instructions here: [Buster [...]

Okay i installed it but how do i start the server now? I am new to raspberry pi and i haven't used linux much before.

starbasessd, 2023-07-13:
The setup instructions set it up to run as a service, auto starting on power up.
To access it, you use http://ip_address_of_Pi:8765. My backup server is at http://192.168.32.90:8765
It works with almost any browser in Windows, Linux, or MacOS and most Android and iPhone browsers.

averagenokiauser, 2023-07-14:
> The setup instructions set it up to run as a service, auto starting on power up. To access it, you use http://ip_address_of_Pi:8765. My backup server is at http://192.168.32.90:8765 It works with almost any browser in Windows, Linux, or MacOS and most Android and iPhone browsers.

Thanks, it works. Is there a way to make it not start when raspberry pi os boots up? I will not always use it as a camera so it would be great if i could just manually start and stop it.

starbasessd, 2023-07-14:
disable auto-start: sudo systemctl disable motioneye
start :  sudo systemctl start motioneye
stop:  sudo systemctl stop motioneye

averagenokiauser, 2023-07-15:
> disable auto-start: sudo systemctl disable motioneye start : sudo systemctl start motioneye stop: sudo systemctl stop motioneye

Thank you so much. Now everything works.

closed 2023-07-15 by averagenokiauser