motioneye-project/motioneyeos #3003: Different Versions, Different Anomolies

Dewey3, 2023-07-22:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190911 & dev20201026

### Board Model

I am using the following board/model: RPi 3A+, 3B, & 3B+

### Camera

I am using the following type of camera: V4L2

My camera model is: ELP 1megapixel Day Night Vision Indoor&Outdoor CCTV USB

### Network Connection

My motionEyeOS unit is connected to the network via: Wifi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

N/A

### Log Files

I consider the following log files relevant to this issue:

N/A

Greetings all!  I am trying to figure out the lesser of two evils, or so to say, the lesser of two versions.  I run 16 RPi USB cams into a 17th headless RPi that serves as the "Console".  This allows us to remote into the Console while home or away and see all 16 cams simultaneously.  The cams store all motion locally and every four hours, my Debian 11 server runs a crontab that reaches out via SSH to each cam (sequentially) and copies (rsync) the stored images to my server's data drive.

I've been running MEOS 20190911 since its release and pretty much stuck to it since it was the most problem free for my system.  The only problem I have with MEOS 20190911 is if someone is viewing the system remotely via the Console, the cams that are catching motion (event) show that event via the Console without a problem, but most times, the event will not be saved locally on the cam.  Because of this, I started playing with MEOS dev20201026 a few months ago.  (I would like to stay with the "older" versions of MEOS).

The problem I have recently been seeing with MEOS dev20201026 is that if the continuity of the system is interrupted by something like a router reboot or quick storm brownout, when everything comes back online **some** of the cams running dev20201026 will refuse all connections, HTTP from the Console AND ssh from the server until I reboot the affected cams.
 
I think I am ready to go back to MEOS 20190911 as I feel it is the lesser of the evils since the only thing that could be lost is the saving of events when the event is being watched live, verses MEOS dev20201026 which could lose EVERYTHING due to system lockup from a router reboot, or momentary story brownout (more likely) when we are away, and until we return.  Before I throw the baby out with the bathwater, I would like to ask if anyone is familiar with what I have described for MEOS 20190911 **or** MEOS dev20201026, and if so, knows of a possible solution.  Thanks all.

starbasessd, 2023-07-22:
How are the cameras getting their IP addresses?
Are they all on WiFi or Ethernet?

Dewey3, 2023-07-22:
> How are the cameras getting their IP addresses? Are they all on WiFi or Ethernet?

All cams are wifi, and the Console is Ethernet.  Everything is hard coded via static_ip.conf to its own reserved IP address.

starbasessd, 2023-07-22:
I'm on Debian 12 currently on my mEye hub/server and don't see that issue. My router (A/B/G/N/AC) is set to reboot every evening, and I don't see those symptoms on my PiZeroWs. I would start looking at the logs, noting the time of the freeze or lockup. I would look at all 5 from motioneyeOS to start.

Dewey3, 2023-07-23:
Thanks Kevin.  I didn't realize that 12 was out, but found out while upgrading as part of my troubleshooting prior to your last reply.  I may have misspoke when I said the Console <> Cam HTTP connections were failing, but I **KNOW** that I could no longer ssh or rsync from the server which put older images at risk of being lost if I was out of town and not available to reboot the affected cams.  Oddly enough while upgrading, the below attachment scrolled up while comparing which visudo I wanted to use, mine or the new version.  I selected the new version and will re-write my specific lines from what I added to the prior version.  Sounds like, and I sure hope the problem I was having was with an update that is now fixed... we'll see.  Thanks again.

[image]

Dewey3, 2023-09-05:
Well, thanks to all, I'm closing this issue.  After searching and searching the net, everything I saw said it was not a "machine" problem, but more than likely a network communication mis-configuring.  Everything I saw referred to a networking level 2 issue and most likely, an ARP (Advanced Resolution Protocal) problem.  So I added a line to my server restart file that runs anytime the server is restarted to clear the ARP table cache (`ip -s -s neigh flush all`), and now I can't force the problem as much as I try.  The only thing I would love to know is why did it only happen randomly with my dev20201026 RPis and not any of my 20190911 RPis?  Speaking of 20190911 -vs- dev20201026, I now have another problem but since it's not related to this, I open another thread.

closed 2023-09-05 by Dewey3