motioneye-project/motioneyeos #3006: Remote Connection Error

Dewey3, 2023-09-05:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: dev20201026 on 16 RPi cameras and ME 0.42.1 on a headless remote console.

### Board Model

I am using the following board/model:  RPi 3B, 3B+, and 3A+.

### Camera

I am using the following type of camera:  V4L2.

My camera model is:  ELP 1megapixel Day Night Vision Indoor & Outdoor USB Camera.

### Network Connection

My motionEyeOS unit is connected to the network via:  all cameras are wifi and the "console" is ethernet.

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 N/A

### Log Files

I consider the following log files relevant to this issue:

 N/A
 
I run 16 RPi cameras that are viewed and controlled by a headless "Console" (#17).  I have migrated all but 3 of my cameras from MEOS 20190911 to MEOS dev20201026.  When running MEOS 20190911, I was able to change the settings on each camera from the "Video Device" selection and below from the Console.  The Console is running ME 0.42.1 on Debian because I wanted tighter control over the Console since it is the connection point for all of the cameras.  I noticed as I have upgraded each camera from MEOS 20190911 to MEOS dev20201026 that I can no longer control the setting on the cameras that are running MEOS dev20201026 from the Console.  To change the settings on any of the MEOS dev20201026 cameras, I have to login directly to the camera that I want to change.  This is not a problem 95% of the time, but an example of where it is a problem is if we are out of town and a spider has built a web over the camera, I have to deal with the thousands of nightly false positives until we return home because I cannot up the sensitivity level of the camera(s) with the spider web(s) from the Console.  When trying to change the camera settings on any MEOS dev20201026 camera from the Console, I always get "An error occurred.  Refreshing is recommended".  Is anyone familiar with this are a way around it, or less desired, but if I drop the Console down to MEOS dev20201026, will I regain remote control of the cameras?  Thanks all.

[image]

starbasessd, 2023-09-05:
@Dewey3 I would bet it's not 0.42.1 on the Debian Console, as I have multiple Consoles connecting to 6 or more cameras, but they are all Python2.7 machines, and they are really 0.42.1s using Python2.7. If your console is a recent install, I'll bet it's the Python3.x version, even if it does say 0.42.1, depending on which instruction set you followed for install and the devs haven't noted it, or updated the version number.

Dewey3, 2023-09-05:
Thank you sir!  So are you saying if I do a Python 2.7 install it should work?  I'm asking because I just look at my instructions that I used for the install and found:

`sudo curl https://bootstrap.pypa.io/pip/2.7/get-pip.py --output get-pip.py`

which leads me to believe that it is Python 2.7.

starbasessd, 2023-09-05:
If it is working as well as you say, I think you are in the Python3 beta test instead of the Python2.7 stable. I would say at this point 'it can't hurt' but you might want to stir up someone like @MichaIng or one of the other devs to see what they think...