motioneye-project/motioneyeos #3007: Does not boot on Banana PI M1

TheFax, 2023-09-27:
### Board Model

Banana pi M1

MotionEyeOS version [20200606](https://github.com/ccrisan/motioneyeos/releases/download/20200606/motioneyeos-bananapim1-20200606.img.xz) does not work at all on Banana pi M1. It does not boot and never reaches the login prompt.

Version 20170326  works quite well, but it is 3 years old.

### Log Files

I cannot get log files. I only read the screen log that appears during startup.

starbasessd, 2023-09-27:
Did you try the dev20201026 [here](https://github.com/motioneye-project/motioneyeos/releases/download/nightly-dev/motioneyeos-bananapim1-dev20201026.img.xz)?
20170326 is 6+ years old. 20200606 had a number of issues across many SBCs. You could also try installing an OS, (preferably the Lite version) and install the motionEye following the instructions [here](https://github.com/motioneye-project/motioneye/wiki/Installation).

TheFax, 2023-09-29:
Yes, I tried today.
It does not work, and it seems to me that the errors are quite similar.
I also took some photos of the monitor in order to help the developers understand the issue.

These photos have been taken during first boot.
First "errors" appears in this screenshot:
[image]

...and after this, the screen is quickly fullfilled of these lines:
[image]

If I wait some minutes, the software seems to try anyway to start-up MotionOs, but it results in a error:
[image]

Like I wronte on the first mesage, login prompt is never reached.
Banana Pi asks correcty an IP via DHCP, but on port 80 there isn't any service active.

[image]

[image]

starbasessd, 2023-09-29:
This is a clean install? Where did you set up SMB services? SAMBA (SMB) is not running by default in motionEyeOS, it must be enabled in the WebGUI, Settings, Services.
If it is enabled and you had never  connected to it, then the SDCard was never properly erased.
Can you SSH into the Banana?

TheFax, 2023-10-01:
Yes, this is a clean install, and what you see is the first boot. This is visible by the line "creating data partition" because this action is performed once at first boot.
I never set-up SMB services because I never reached WebGUI.
It is not possible to SSH into the Banana.

I tried with 2 different SD cards, both 4GB, and never deleted them because writing the image is by itself a deletion of original content.
I used 3 different way to burn them:
1) WIndows - With **[Rufus](https://rufus.ie/it/)**
2) Windows - With **[Win32 Disk Imager](https://win32diskimager.org/)**
3) Linux - Using official **[writeimage.sh script](https://raw.githubusercontent.com/ccrisan/motioneyeos/master/writeimage.sh)**

Despite this, I cannot ensure SD cards has issues. Version 20170326 works using same hardware and same burning software.

TheFax, 2023-11-30:
Today, I performed the test using a brand new 16GB SD card. The results were the same as those obtained in my [second post](https://github.com/motioneye-project/motioneyeos/issues/3007#issuecomment-1740462025).
I also tried downloading the 20200203 version, which exhibited the same behavior.

I read the scripts that runs on first boot: I'm not able to see the starting of "overlay_fs" in my screen logs.

starbasessd, 2023-11-30:
I would try another OS install that supports Python 2.7 (Debian Buster does) because I doubt the new devs are probably uninterested in supporting the older SBCs that would require their own images.
Other OS install instructions [here](https://github.com/motioneye-project/motioneye/wiki/Installation)