motioneye-project/motioneyeos #3009: Cannot get *supposedly supported* USB wifi adapter to work

derekantrican, 2023-10-02:
Similar existing issues that I have found: #2065, #900, #1663

### motionEyeOS Version

motionEye Version: 0.42.1
Motion Version: 4.2.2+gitUNKNOWN
OS Version: motionEyeOS 20200606

### Board Model

Raspberry Pi 1

### Camera

I'm using this camera: https://labists.com/products/raspberry-pi-webcam-rpi-camera-module-official (though that's probably not relevant for this issue)

### Network Connection

I am attempting to connect via this USB wifi adapter: https://thepihut.com/products/usb-wifi-adapter-for-the-raspberry-pi which - according to that page - is a MediaTek MT7601U chipset. I chose that wifi adapter because [this comment](https://github.com/motioneye-project/motioneyeos/issues/2065#issuecomment-525472171) suggested I check the list of thingos list of drivers. While the links in that comment are outdated, I believe that chipset is supported by looking here: https://github.com/ccrisan/thingos/blob/main/configs/raspberrypi_defconfig#L35

However, this does not work. When booting up the pi with a screen attached, USB wifi dongle attached, and ethernet NOT attached, the following message is on the screen:
> Preparing AP interfaces: none
Starting wpa_supplicant: no device
Configuring wireless network: no link
anic: rebooting (caused by network)

And of course, the pi reboots repeatedly until I plug in an ethernet cable connected to my network.

### Peripherals

I am using the following peripherals that I consider relevant to this issue: (USB wifi adapter linked above)

### Log Files

I consider the following log files relevant to this issue:

[motioneye.log](https://github.com/motioneye-project/motioneyeos/files/12779179/motioneye.log)
[motion.log](https://github.com/motioneye-project/motioneyeos/files/12779180/motion.log)
[dmesg.log](https://github.com/motioneye-project/motioneyeos/files/12779181/dmesg.log)
[boot.log](https://github.com/motioneye-project/motioneyeos/files/12779182/boot.log)
[messages.log](https://github.com/motioneye-project/motioneyeos/files/12779183/messages.log)

I am not sure if this is visible in the above logs, but for what it's worth, I've also configured my wireless network settings via the UI:
 
[image]

starbasessd, 2023-10-02:
Are you using any other USB devices? Maybe a (un)powered USB hub?
Have you tested the dongle on the Pi1 with a Legacy (Buster) install to confirm it's working?
Have you tried [dev20201026](https://github.com/motioneye-project/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi-dev20201026.img.xz)?

derekantrican, 2023-10-08:
Sorry for the delay - a lot on my plate right now.

No, I am not using any other USB devices. This is the only USB device plugged into the pi.

I will try the other suggestions & report back

starbasessd, 2023-10-08:
Looking in dmesg, it is at least discovered:
[    2.223555] usb 1-1.3: new high-speed USB device number 4 using dwc_otg
[    2.365834] usb 1-1.3: New USB device found, idVendor=148f, idProduct=7601, bcdDevice= 0.00
[    2.365860] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.365875] usb 1-1.3: Product: 802.11 n WLAN
[    2.365887] usb 1-1.3: Manufacturer: MediaTek
[    2.365898] usb 1-1.3: SerialNumber: 1.0

Will try to dig out a Pi1 or a PiZero without an on-board WiFi chip and test...

starbasessd, 2023-10-08:
I couldn't find a Pi1, but I have several PiZeros (uses the same image) that do not have the Wifi chip (PiZero V1.3). I installed motioneyeos-raspberrypi-dev20201026.img, and added the files ssh.txt and wpasupplicant.conf to the boot partition. 
I have an hdmi capture card, connected it. used a USB type a to type micro b adapter and tried several random wifi N dongles from asus, tenda, and edimax, and all worked. All showed pulling an IP address, time & date in boot.log and on screen. I don't have your wifi available here to test. Most realtek chipsets work, though.

derekantrican, 2023-12-01:
Oh, woah. Just got it to work! I was using the raspberry pi imager but I guess it wasn't placing the wpa_supplicant.conf file. I used Win32 disk imager instead and manually put the file myself.

Can confirm that it booted up just fine and can connect to wifi now! Thanks!