motioneye-project/motioneyeos #3014: Dropping fps after some minutes

Dukadyr, 2023-11-04:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version: motionEyeOS 20200606

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model: Pi Zero w and Pi Zero 2 w

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera : Pi Camera v2

I am using the following type of camera:  Fast Network Camera

### Network Connection: Wifi 2,4GHz/5GHz

My motionEyeOS unit is connected to the network via:  WIFI

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 
Hey folks,
the issue:
my (Fast Network Camera) stream works for some minutes (2-4) fine with 15-20fps, but after a while its dropping to 1-2 fps.
At this moment I'm running following Settings: 800x600 / 15fps / Bitrate: 1000000/ Image Quality: 15%

I have tried all kinds of settings, like changing GPU Memory, Overclocking and Underclocking, changing Quality.

On the Wifi Connection i get 14 / 5 MBit/s Signal.
Logfiles are clean, no Errors...

Is there someone with another Hint or Solution for me?

starbasessd, 2023-11-04:
The PiZero2W would use the Pi3 image or the other image that Jawsper posted (pinned in Issues). it is not supported with the PiZero/Pi1 image.
I suggest shifting the PiZero to dev20201026 image.
Make sure the Frame Rate dimmer and Resolution Dimmer are set to 100.

Dukadyr, 2023-11-04:
> The PiZero2W would use the Pi3 image or the other image that Jawsper posted (pinned in Issues). it is not supported with the PiZero/Pi1 image. I suggest shifting the PiZero to dev20201026 image. Make sure the Frame Rate dimmer and Resolution Dimmer are set to 100.

yes, i know.
the image version i used on the pi zero 2 w was jawspers image.
now I am running the Pi Zero w with motionEyeOS 20200606.

starbasessd, 2023-11-04:
I would also check for thermal throttling using the 
>vcgencmd get_throttled 

command.

I use a script to get human-readable info:

= = = = =

#!/usr/bin/env python2

import subprocess

GET_THROTTLED_CMD = 'vcgencmd get_throttled'
MESSAGES = {
    0: 'Under-voltage!',
    1: 'ARM frequency capped!',
    2: 'Currently throttled!',
    3: 'Soft temperature limit active',
    16: 'Under-voltage has occurred since last reboot.',
    17: 'Throttling has occurred since last reboot.',
    18: 'ARM frequency capped has occurred since last reboot.',
    19: 'Soft temperature limit has occurred'
}

print("Checking for throttling issues since last reboot...")

throttled_output = subprocess.check_output(GET_THROTTLED_CMD, shell=True)
throttled_binary = bin(int(throttled_output.split('=')[1], 0))

warnings = 0
for position, message in MESSAGES.iteritems():
    # Check for the binary digits to be "on" for each warning message
    if len(throttled_binary) > position and throttled_binary[0 - position - 1] == '1':
        print(message)
        warnings += 1

if warnings == 0:
    print("Looking good!")
else:
    print("Houston, we may have a problem!")
= = = = =
[oops.py.txt](https://github.com/motioneye-project/motioneyeos/files/13256696/oops.py.txt)

Dukadyr, 2023-11-04:
> I would also check for thermal throttling using the
>
>
> command.
>
> I use a script to get human-readable info:
>
> = = = = =
>
> #!/usr/bin/env python2
>
> import subprocess
>
> GET_THROTTLED_CMD = 'vcgencmd [...]

Thx for the Hint!

Checking for throttling issues since last reboot...
Looking good!

starbasessd, 2023-11-04:
I would then switch to the dev20201026 image, and check the Dimmer settings.

Dukadyr, 2023-11-04:
> I would then switch to the dev20201026 image, and check the Dimmer settings.

Dimmers are on 100%
Ill give the dev image a try. Thx bud!

kRYstall9, 2024-09-16:
@Dukadyr Did you fix your problem with the dev20201026 pre-release?