motioneye-project/motioneyeos #3015: Connectivity Watch Interval ... A Question about the Interval

Dewey3, 2023-11-10:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: dev20201026

### Board Model

I am using the following board/model: RPi 3A+, 3B, and 3B+

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: ELP 1megapixel Day Night Vision Indoor & Outdoor CCTV USB Dome Housing Camera

### Network Connection

My motionEyeOS unit is connected to the network via: Wifi

### Peripherals

I am using the following peripherals that I consider relevant to this issue: N/A

### Log Files

I consider the following log files relevant to this issue: N/A

I have been using MEOS for YEARS and I have no idea why I am just finding and realizing this fix.  Every once in a while... 1 to 3 times a month, some of my cameras will randomly loose connectivity to my server and no longer respond.  I cannot SSH into the "lost" camera get into the "lost" camera via the console HTTP connection, which requires a physical and manual reboot of the lost camera.  The lost camera itself will still function and capture and store images, I just can't connect to the lost camera(s).  I've always used the "Network Link Watch", but that would not resolve the problem when it would occur.  Anyway, I am just recently realizing that "Connectivity Watch" is just what the doctor ordered and has already proven its worth a couple of times.  I have set Connectivity Watch to my home router, 192.168.1.1, and now when the problem has happened, the affected cameras will reboot and come back online after the Connectivity Watch parameters have been met.  This has worked so well, I've turned "Network Link Watch" off since Connectivity Watch appears to basically do the same thing, just a different way.  My question is about "Connectivity Watch Interval".  Is the "interval" the amount of time between **every** attempted Connectivity Watch, or just the amount of time between the attempted connections when Connectivity Watch **fails** on the first attempt?

Dewey3, 2023-11-10:
For the sake of anyone who comes upon this thread, it looks like viewing the logs answers my question.  I had the Connectivity Watch Retries set to 3 and the Connectivity Watch Interval set to 3600 (1 hour).  Looking at the Messages log, it looks like after the third **failed** attempt in the third hour that my camera rebooted:

```
Nov  4 01:02:42 Driveway user.notice netwatch: cannot connect to 192.168.1.1:80
Nov  4 02:02:51 Driveway user.notice netwatch: cannot connect to 192.168.1.1:80
Nov  4 03:03:00 Driveway user.notice netwatch: cannot connect to 192.168.1.1:80, calling panic action
```

The day that this occurred, I noticed that my server, which makes hourly connections to check every RPi I have took 3 hours before it rebooted and the connection was restored, so I changed the Connectivity Watch Reties to 2 so the affected camera would not be out of contact for 3 hours.  This then happened again a few days later, but this time it took only 2 hours for the camera to reboot:
```
Nov  9 21:29:10 Driveway user.notice netwatch: cannot connect to 192.168.1.1:80
Nov  9 22:29:19 Driveway user.notice netwatch: cannot connect to 192.168.1.1:80, calling panic action
```

So with that I can safely see that the Connectivity Watch Interval is the number of times between **failed** connections before the affected camera will reboot.

closed 2023-11-10 by Dewey3

starbasessd, 2023-11-10:
Very interesting, @Dewey3 Thank you for your own research. I am always looking for watchdog type information, and I will try to look into it this weekend, myself, too.