motioneye-project/motioneyeos #3018: Error opening camera = Unable to open video device

cohen-sydney, 2023-12-14:
I am getting this error on the connection of the camera.
the camera does work I have proved it on the same hardware with libcamera I get a video feed.

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motionEyeOS 20200606 // 0.42.1 // 4.2.2+gitUNKNOWN

### Board Model

Rpi Zero w

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera:  V4L2

My camera model is: (insert your camera model, e.g. Logitech C290).

RPI Camera 3 == direct connected to the board via a ribbon 

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

Wifi 

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

None

### Log Files

I consider the following log files relevant to this issue:

[motioneye-5.log](https://github.com/motioneye-project/motioneyeos/files/13679089/motioneye-5.log)

[boot.log](https://github.com/motioneye-project/motioneyeos/files/13679091/boot.log)

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 
thanks for any help.

cohen-sydney, 2023-12-14:
I ran this command
vcgencmd get_camera

supported=1 detected=0

starbasessd, 2023-12-14:
Camera V3 is not supported directly in motionEyeOS ( [even dev20201026](https://github.com/motioneye-project/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi-dev20201026.img.xz) ) because it was released after the last image was published, and the new devs have not moved forward with a new image. 
Some people with 'advanced skills' in RaspberryPiOS have had some success in replacing the driver files and kernel and getting it to work. 
Your best bet if you want to use it with motionEye would be to install RPiOS Buster and then following the [instructions here](https://github.com/motioneye-project/motioneye/wiki/Install-On-Raspbian)

cohen-sydney, 2023-12-14:
thanks. appreciate the quick reply.

starbasessd, 2023-12-14:
Be sure to NOT use the latest RPiOS (Bookworm) as it does NOT support Python2 or the needed APIs that motionEye uses...

cohen-sydney, 2023-12-14:
Would Bullseye also work?? as that is simple to install from he RPI imager

starbasessd, 2023-12-14:
Not if you want to use a Camera V3. USB, Yes. Buster has all the APIs and Python2.7 available.

bibiqt40, 2024-01-01:
Hey @cohen-sydney ,

Did you have any luck with this one? I'm in a similar boat, I got a Raspberry Pi Zero 2 with the v3 Wide Camera plugged in with the adaptor cable.

When I get MotionEyeOS running on the board, the camera always says "Unable to open video device. Were you able to get your set up running?