motioneye-project/motioneyeos #3019: RPI ip not appearing in the network list

DuduCreator, 2023-12-22:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606

### Board Model

I am using the following board/model: raspberry pi 4 

### Camera

i dont consider this relevant

### Network Connection

My motionEyeOS unit is connected to the network via: WIFI.

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

64 GB micro sd card

### Notes

This is my first time running motioneyeos on RPI 4 headless. When im plugging it its not appearing on my network so im not able to connect to it.

I flashed my image with the RPI imager, with no custom os settings and added `wpa_supplicant.conf` file with the template from the wiki.

```
country=RO
        update_config=1
        ctrl_interface=/var/run/wpa_supplicant
        
        network={
            scan_ssid=1
            ssid="MyWiFiSSID"
            psk="S3cr3tp@$$w0rc|"
        }
```
Obviously i added the correct ssid and password.

starbasessd, 2023-12-22:
My wpa_supplicant.conf file format:
- - - - -
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
    ssid="my_ssid"
    psk=encrypted_psk
}
- - - - -
Have you hooked up a monitor to see what is going on? There are issues that might put it into a continuous boot loop. 
1) bad format of the wpa_supplicant.conf file
2) using RPiImager, as it does not handle the 3 partitions well. I recommend Win32DiskImager, or, cautiously, Balena Etcher.
3) If it is not booting completely, and you can attach a monitor, you can see where it is failing. At the rainbow screen indicates a bootloader issue, as does the specific message you need newer software) or the bootloop issue which is the bad format of the wpa_supplicant.conf file.
4) A lot of Pi4s have an issue with the bootloader. If it is not in the range near 4/2020, it will fail, especially if newer (like late 2021 or newer)
5 for the Pi4, I also recommend dev20201026 as it is much more stable on both my Pi4-4GB and Pi4-8GB.

DuduCreator, 2023-12-22:
Currently i dont have with me a micro hdmi cable, so i am not able to hook up a monitor. I will try using the Win32DiskImager, go to the dev20201026 version and modify the `wpa_supplicant.conf ` file. Is it a way to see what's happening there without a monitor?

starbasessd, 2023-12-22:
If you have access to your router's DHCP server to check the lease, that would give you the IP address. Oh, that reminds me, do you add the ssh or ssh.txt file (nothing in it) to the same partition as wpa_supplicant.conf file? That is needed just like on a regular raspberry pi os.
Another thing you could try: Install RaspberryPi OS to an SDCard, with the wpa_supplicant.conf file and ssh or ssh.txt file or set it up using the extra settings in RPiImager, and find the IP address for it. Then swap out to the motionEye/motionEyeOS SDCard and try the same IP Address.

DuduCreator, 2023-12-22:
I use Fing to check for my RPI, and i have the ip saved in the app. It shows me "raspberry pi (ip adress) - Offline"

This time i used Win32DiskIager, with the dev version and it still doesnt appear to be online or if i have a new device on my network. 

Maybe i put the files in the wrong directory
[image]

When i switched from raspberryPi OS docker to motioneyeos (i had some problems with docker) i tried the same ip address that i've known and it said that is couldnt reach the ip adress.

DuduCreator, 2023-12-22:
I have also entered in my router and tried all the unknown ip adresses. You just write the ip adress in a browser, without any ports, am I right? I dont want to make a dumb mistake and take up your time and mine.

starbasessd, 2023-12-22:
Looks right in your image.I would bet you are getting the rainbow screen, see 3 and 4 above, probably the bootloader issue. If a normal RPiOS image boots, you can run the
>vcgencmd bootloader_version 

and report the date. Hard to determine without a hdmi cable as to which issue it is.

starbasessd, 2023-12-22:
Do you have an ethernet cable and test if it boots up with that?

DuduCreator, 2023-12-22:
Yeah I have it but it will take some time. Do I have to delete the `ssh` and `wpa_supplicant.conf` file when using a wired connection?

starbasessd, 2023-12-22:
If the ssh file is there, I'd try PuTTY or other SSH client, too.
For motionEyeOS you may have to use 
> http://ip_address  

specifying http:// as most browsers today default to https://

starbasessd, 2023-12-22:
> Yeah I have it but it will take some time. Do I have to delete the `ssh` and `wpa_supplicant.conf` file when using a wired connection?

No, you can leave them in place.

DuduCreator, 2023-12-22:
I am a bit confused, in my router, it appears a checkmark next to the LAN 1 (raspberry pi) field and says "ip checked", but when i go into the attached devices list i cant find any new device. 

Should i go with the version on the RaspberryPI OS ? Since RaspberryPi OS worked for me when i was using docker.

starbasessd, 2023-12-22:
Depends. 
You can't use the latest RPiOS (bookworm) at all, it doesn't have Python2.7 available. 
You can use Bullseye if you are not using CSI ribbon cable cameras. USB and Network cameras work fine, though.
You need Buster if you are using CSI Ribbon Cable cameras. 
Buster and Bullseye have Python2.7
Bullseye does NOT have the version of the CSI cameras APIs.
Buster and Bullseye can be found here: https://www.raspbian.org/RaspbianMirrors/
find a close site, and drill down in dists, and go for Buster or Bullseye as needed
Instructions to install are [HERE](https://github.com/motioneye-project/motioneye/wiki/Installation)
Just don't want you to wander off into hair pulling land...

starbasessd, 2023-12-22:
One other thing:  If you install motionEye on another OS (like RaspberryPiOS) you would then need to call it as http://_ipaddress_:8765 as it doesn't run on the 'normal' web port (80)

DuduCreator, 2023-12-22:
I think i will wait and get a hdmi cable when its possible to (probably after New Year) and I will come back or I will go with the RPI OS, but now its too late in my time zone. 

When I come back, do I open a new issue?

Thanks for helping me out and being so fast. You guys are really good on problem solving (from my expierence with others). 

And also Bulllseye isnt a problem for me because im using an USB camera.

starbasessd, 2023-12-22:
Yes, I would close this one, and open a new issue in the [motionEye issues github](https://github.com/motioneye-project/motioneye/issues).

closed 2023-12-23 by DuduCreator