motioneye-project/motioneyeos #3020: VNC with MotioneyesOS

TannarV, 2023-12-27:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (motioneyeos-raspberrypi3.img).

### Board Model

I am using the following board/model: (Raspberry pi zero 2 w).

### Camera

I am using the following type of camera: (Spy Cam V1).

My camera model is: (SpyCam).

### Network Connection

My motionEyeOS unit is connected to the network via: (2.4Ghz Wifi).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (micro USB)

### Log Files

I have successfully connected to motion eyes, however i am no longer able to connect to my vnc server cuz i have no monitor to plug it into. Is it possible to run a vnc server/ raspi os and motioneyes os.

My goal is to make a rocket with a camera in it therefore i need to be able to code the flight software on the pi while also having access to the camera.

My ideal outcome is to eventually make some sort of app/webbrowser to see flight data and the camera during a flight.

Any help is greatly appricated!

starbasessd, 2023-12-27:
MotionEyeOS runs a web service GUI at port 80. It does not have the resources to run a VNC server, there isn't an X11 server. 
Interesting use case. 
I would run a bare minimum Debian 11, install X11, install a VNC server, and install motionEye. MotionEye's web service is on port 8765 instead of 80 like motionEyeOS. I would enable a WiFi access point, too.  This would allow you to run almost anything else for data, Python3, nodered, java, whatever, and with a web gui or vnc access.

TannarV, 2023-12-27:
Thank you very much for this. I dont know what a lot of this is but i will research and try it out. Thanks for the help!

starbasessd, 2023-12-27:
MotionEyeOS is self contained and can't run much else.
Debian 11 desktop for RaspberryPi: https://raspi.debian.net/
RaspberryPiOS that can support motionEye: https://downloads.raspberrypi.org/raspios_oldstable_armhf/images/raspios_oldstable_armhf-2023-12-06/2023-12-05-raspios-bullseye-armhf.img.xz
RaspiOS has VNC Server built in (enable in raspi-config)
I don't know what kind of data or how you want to collect it, that is what Python3, nodered, java and other languages are used for.
To talk to the Raspberry Pi, you'd probably use wifi, set it up as an access point host...
If you need help or want suggestions on how to get it all organized, let me know.