motioneye-project/motioneyeos #3022: Camera via USB randomly stops being detected, adding and deleting new camera temporarily works

sts-ryan-holton, 2024-01-07:
### motionEyeOS Version

I am running motionEyeOS version: motionEyeOS dev20201026

### Board Model

Raspberry Pi 4

### Camera

A simple USB camera, [this one](https://www.amazon.co.uk/gp/product/B089LTY8X7/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1)

### Network Connection

Wi-Fi, within 10 metres of the router

### Peripherals

Just the single USB camera

### My problem

Okay, so I've been running this USB camera attached to my Pi on this version for well over a year, however, within the past few months I've noticed randomly the camera just stops getting picked up and doesn't record. I've found if I add the same camera again, and delete the camera I just added, the original camera setup will suddenly start working again, but stops working after several minutes. What's going on here and how could I resolve this?

However, if I keep doing this, I get the unable to open video device randomly too

[image]
[image]
[image]
[image]

### Log Files

I consider the following log files relevant to this issue:
 
[motioneye.log](https://github.com/motioneye-project/motioneyeos/files/13853465/motioneye.log)
[boot.log](https://github.com/motioneye-project/motioneyeos/files/13853464/boot.log)

starbasessd, 2024-01-07:
Going through your motioneye log, you are also having issues mounting your smb share.
Going through your boot log, I see it rebooting every few hours without doing a reboot command (no normal reboot shutdown / restart.
A couple of things I'd look at and consider:
1) Power supply issues. Is it 'borderline' on how much power it can supply to the Pi and through to the USB Camera? Do you have another power supply you can test with?
1a) Are there any warnings in the logs? Can you post the messages and dmesg logs, please?
2)  Do you have the camera plugged into a blue jack or black jack on the Pi? Can you switch to the other color? (Blue is USB3, black is USB2, and they use different chips on the board)
3) Do you have another computer (PC preferred) that you can test the camera with for a bit, o see if it is the problem?

sts-ryan-holton, 2024-01-07:
Hi,

The rebooting will be down to the fact that I have a wireless wifi plug that is set on a schedule to turn off every few hours for a few minutes and turn back on. I actually added this to the Pi because the camera output would freeze periodically and restarting the Pi actually resolved that.

As for the network share mount, interesting because it doesn't connect to my synology anymore? Remember, I haven't changed these settings in MotionEye OS for months, and neither have I changed the Synology. I checks, SMB is turned on and set to SMB v2 on my NAS, yet MotionEye OS can't connect?

Furthermore, yes, the USB camera doesn't work right now in either ports, yet works fine in another PC

[image]

starbasessd, 2024-01-07:
1) What happens when you click the Test Share button?
2) Can you post the messages and dmesg logs, please?

sts-ryan-holton, 2024-01-07:
Right, I've resolved the network share problem. it was caused by the IP address changing, I've changed the IP. When I did this the camera started recording successfully and uploading successfully to the share. It's been around 5 minutes since it's been working and now it's stopped again. 

Here's the additional logs now:

[messages (1).log](https://github.com/motioneye-project/motioneyeos/files/13854091/messages.1.log)
[boot (1).log](https://github.com/motioneye-project/motioneyeos/files/13854095/boot.1.log)
[dmesg.log](https://github.com/motioneye-project/motioneyeos/files/13854096/dmesg.log)
[motion.log](https://github.com/motioneye-project/motioneyeos/files/13854099/motion.log)

And the files recorded before it stopped:

[image]

starbasessd, 2024-01-07:
I would do the following:
1) Make a backup of the camera & system setups by going into Settings, General Settings, Configuration, Backup 
2) Reimage the SDCard with dev20201026. 
3) restore the backup by going to Settings, General Settings, Configuration, Restore 
4) confirm the network connection to the NAS is working.

Watch for the issue. It is possible the power off resets caused corruption.
If the problem reduces, I can give you a cronjob to have the motionEyeOS reboot cleanly however often you want. If it goes away (like days instead of hours) we can still use the reboot cronjob daily or something...

sts-ryan-holton, 2024-01-07:
Okay,

I think I've resolved it. All I did was move the USB camera over to the USB 3.0 (blue) port and update the IP, so to clarify what resolved it for me:

1. Update the network share IP as it had changed.
2. Move the USB camera to USB 3.0 (blue) port as the DMESG suggested a potential lack of power. Googling shows that USB 3.0 can provide more power to a device.

I'll close this now, thanks for your help!

closed 2024-01-07 by sts-ryan-holton

starbasessd, 2024-01-07:
Current supplied to the USB Ports is whatever your power supply gives minus about 2.5 amps used by the Pi. RPiFoundation says using their wall wart allows about 0.5 amps. If you have anything else plugged in like keyboard, mouse, monitor, etc, they all share that power. So does the ethernet connection and wifi connection. The wifi also shares the USB bandwidth, as does the ethernet.