motioneye-project/motioneyeos #3025: Wifi Config Issue

tejasthy, 2024-01-28:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606.

### Board Model

I am using the following board/model: Rasberry Pi Zero W.

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: Wifi.

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 
The wifi config file is not working. I've attached it correctly as detailed in the instructions but still not working. I am using an enterprise wifi (in a dorm). Also for some reason it keeps trying to connect to a wired network even though there is no wired network. Thanks.

starbasessd, 2024-01-28:
MotionEyeOS does not support WPA3, nor does it support some passwords with some 'special' characters. Here is an example of how your wpa_suplicant.conf file should look:
= = = = =
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
    ssid="SSID"
    psk="Secr3T_p@ssW0Rd"
}
= = = = =
Of course, supply your needed ssid in place of SSID and your secret password in place of Secr3T_p@ssW0Rd and your proper country code for US.

How are you creating the wpa_supplicant.conf file? Windows, Linux, or Mac?
Are you putting it in the /boot folder prior to first boot?
Did you make sure that it is named wpa_supplicant.conf, and not wpa_supplicant.conf.txt if creating it in Windows?

Another option may be to configure the PiZeroW as an access point and connect to it directly rather than through the dorm Wifi. If you are interested in this option, I can post the files needed that you can modify and add.

Also, on the back of the Pi is some text in white. It says 
"Raspberry Pi Zero W V1.1" If it says "V1.3" or something else, you have a non-wifi capable PiZero.

tejasthy, 2024-01-28:
I fixed the initial problem but now I'm having trouble connecting to the enterprise wifi.
`country=US
update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
	ssid="MWireless"
	scan_ssid=1
	key_mgmt=WPA-EAP
	eap=PEAP
	identity="username"
	password="password"
	ca_cert="/etc/ssl/certs/AddTrust_External_Root.pem"
	phase1="peaplabel=auto peapver=0"
	phase2="auth=MSCHAPV2"
}`

Thanks.

starbasessd, 2024-01-28:
MotionEyeOS does not support your path for ca_cert /etc is part of a read only file system. You would not be able to put that file in there. You might be able to put it in the /data/etc/ folders. 
I do not believe motionEyeOS supports EAP encryption...