motioneye-project/motioneyeos #3026: Unstable IPv4

cz-lucas, 2024-03-03:
Hello,

I have the problem IPv4 is very unstable. I am getting a very high packet loss and pinging the pi fails most of the pi. Also the SSH-connection and the web interface isn't possible due the issue. Pinging the Pi using IPv6 works without any pocketless. Sadly isn't the web interface or ssh available.

I am using a Unifi UAP-AC-LR (Firmware: 6.6.55.15189) as the access point. 
Wifichannel: 11 

[image]
[image]

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314). 20200203

### Board Model
Raspberry Pi Zero W

### Camera
Fast Network Camera

My camera model is: Raspberry pi Camera

### Network Connection

Wifi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:
Nothing is connected to the pi except the camera

### Log Files

I couldn't get log files because of the very unstable ssh-connection.

starbasessd, 2024-03-03:
I would recommend the dev20201026 version as he earlier versions had a number of issues. All 4 of my PiZeroW cameras use the dev20201026 version around the house without issue to the WebGUI and SSH connections on TP-Link TL-SG108E APs in AP mode and WPA2 security in a neighborhood with numerous AT&T, Spectrum, and other stand-alone APs. Also on my APs are  large number of IoT and Home Assistant devices,

starbasessd, 2024-03-03:
Sorry, not enough caffeine. My APs are TP-Link Archer AC1900s, the previous is my test VLAN switches...