motioneye-project/motioneyeos #3028: PI Zero stuck on rainbow screen wifi config file not working

BeTheChange07, 2024-03-14:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.
yes

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.
yes

### motionEyeOS Version

I am running motionEyeOS version: 20201026

### Board Model

I am using the following board/model: Raspberry Pi Zero 2w
### Camera

I am using the following type of camera: Na

My camera model is: na.

### Network Connection

My motionEyeOS unit is connected to the network via: WIFI

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 micro sd car KOOTION 32gb

### Log Files

I consider the following log files relevant to this issue:

na

***********************************************************************************
I have followed the steps outlined here: https://github.com/motioneye-project/motioneyeos/wiki/Wifi-Preconfiguration

i write the 

i am still getting the rainbow screen when turning on the zero for the first time. 

i have had success on my raspberry pi 4 4gb and that is now streaming successfully.

I'm using win32 to flash the image

iv tried everything for days now and i still cant get it to work
 
i have cant find the boot drive (to maybe add it there) when using Linux

i have been flashing from a windows computer

starbasessd, 2024-03-14:
2 questions:
1) which image are you using? You need to use the one[ here](https://github.com/motioneye-project/motioneyeos/issues/2875)
2) what are you using to image the sdcard? Do NOT use RPi Imager, use either Win32DiskImager or Balena Etcher. RPi Imager is not set up for images using 3 partitions like motionEyeOS.

BeTheChange07, 2024-03-14:
[image]

USING THIS WORKED THANK YOU SO MUCH!! 

ps. i would sugest making it the official pi zero release

closed 2024-03-14 by BeTheChange07

starbasessd, 2024-03-14:
The Pi ZeroW and PiZeroW2 are totally different.

BeTheChange07, 2024-03-14:
Ahh i see. Thanks again!