motioneye-project/motioneyeos #3029: Raspberry Pi 4B with HQ camera not showing in MotionEyeOS?

Solarbot, 2024-04-30:
Hi,

Hoping this is a simple one to fix as I'm not very experianced !?

I have my Raspberry Pi 4B and Raspberry HQ camera runing fine on the stock Raspberry Pi OS with MotionEye installed - all good there.  I have then been keen to get the same hardware running MotionEyeOS and have tried both versions below.  To get MotionEyeOS to boot I had to copy  **start4.elf** and **fixup4x.dat** over from the the working SD card and from there all seems good apart from the camera screen is showing "Unable to open video device" but in the VIDEO DEVICE options it seems to be there:

Camera ID: 1
Camera Device: /dev/video10
Camera Type: V4L2 Camera

I have seen other people have had success running the same hardware/software setup so I guess I'm missing a simple step to do with enabling the camera?  In the MotionEye version that works I had to enable the camera in raspi-config but this doesn't seem to be possible in MotionEyeOS or not at least if I type 'sudo raspi-config' from the command prompt? 

Any help or pointers much appreciated :-)

### motionEyeOS Version
motioneyeos-raspberrypi4-20200606.img 
motioneyeos-raspberrypi4-dev20201026.img

### Board Model: Raspberry PI 4B 2GB

### Camera: Raspberry Pi HQ camera V4L2

### Network Connection: Ethernet

starbasessd, 2024-04-30:
The HQ camera was released after the dev20201026 image, and was not supported which is why it was always recommended to do an RPiOS install and motionEye...

starbasessd, 2024-04-30:
Which camera do you have? There have been many versions, using different sensors, and by different mfgs... Does your camera have the official raspberry pi logo and model #?  
When running motionEyeOS (preferably [dev20201026](https://github.com/motioneye-project/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi4-dev20201026.img.xz)) what does the command:
>vcgencmd get_camera

give you?

Solarbot, 2024-04-30:
This is the original Raspberry Pi V1.0 HQ camera dated 2018 on the PCB so in theory it has the Sony IMX477 sensor?  I can't access the board at the moment but will run that command tomorrow and report back.

Thank you.

gigenthomas, 2024-05-19:
Hello @Solarbot - Where did you copy  start4.elf and fixup4x.dat files from ? I am unable to boot MotionEyeOS with Raspberry Pi 4 Model B