motioneye-project/motioneyeos #3030: bypass Motion Detection?

matt0816, 2024-05-04:
Hi,

I'm working on a traffic camera that should record footage from intersections within a defined time period, such as from 6 to 10 a.m. and from 3 to 7 p.m. I want the videos to be split into segments with a length of 900 seconds. I don't actually need motion detection, but I like the performance and user interface of MotionEye. I believe I've figured out that the "working schedule" only affects the "motion-triggered" recording mode. When I set up continuous recording and set a maximum movie length of 900 seconds, MotionEye only records one video of 900 seconds and then stops. As I've read in some other posts, this is a known issue?

For the given example, it would be nice if I could record the time period from 6 a.m. to 7 p.m. in "motion-triggered" recording mode and bypass the motion detection through some suitable options so that MotionEye always detects motion and the recording doesn't stop until the maximum length is reached, and then starts a new video. Is this possible?

It would be nice if someone could guide me in the right direction.

Thank you very much.

my Setup:
I am running motionEyeOS version: motionEyeOS 20200606
I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).
I am using the following type of camera: MMAL

Marijn0, 2024-05-04:
I'm no expert but I use:
/usr/bin/curl "http://localhost:7999/1/config/set?emulate_motion=1"
to turn on a motion event and:
/usr/bin/curl "http://localhost:7999/1/config/set?emulate_motion=0" to end it. that is also mentioned here: https://github.com/motioneye-project/motioneye/issues/229#issuecomment-653850018

You could use a cron job to do this automatically:
crontab -e

0 6 * * * /usr/bin/curl "http://localhost:7999/1/config/set?emulate_motion=1"

0 10 * * * /usr/bin/curl "http://localhost:7999/1/config/set?emulate_motion=0"

0 13 * * * /usr/bin/curl "http://localhost:7999/1/config/set?emulate_motion=1"

0 19 * * * /usr/bin/curl "http://localhost:7999/1/config/set?emulate_motion=0"

And you may need to set the Working Schedule so that you don't record outside of these times.

matt0816, 2024-05-05:
Thanks for the reply. This helps me, but I noticed two things:
1. if I set the entry at 2:30 p.m. to 1, the recording starts even if the working schedule is set between 3 and 4 p.m.
2. After reaching the maximum video length of 900 seconds, the motion detection is automatically set back to 0 and no further recording is started.

Okay, I'll ignore the second point for now, then the videos will be 4 hours long. But for ease of use by less experienced users, it would be great if the times after changing the working schedule or after clicking "Apply" could be written into the crontab using a separate script. To do this, I would have to find a way to start the script after clicking "Apply" and I would have to know where the working schedules are stored.

Does anyone here have an idea?