motioneye-project/motioneyeos #3031: Issue Mounting SMB Share

kevo970, 2024-05-23:
I am running MotionEye from a virtualized Home Assistant OS which is hosted on TrueNAS Scale.

I had the SMB share working normally on a similar setup a few weeks ago (HAOS was running in a VirtualBox VM from my Windows PC and my NAS machine was on TrueNAS Core), but I re-built everything after a hardware upgrade for my NAS.

I have tested the share and account from my Windows desktop and the credentials and IP all work fine, but when I enter them into the MotionEye GUI, my logs spit this out:

mount error(113): could not connect to 192.168.0.200Unable to find suitable address.
mount error(113): could not connect to 192.168.0.200Unable to find suitable address.
mount error(113): could not connect to 192.168.0.200Unable to find suitable address.
mount error(113): could not connect to 192.168.0.200Unable to find suitable address.
   ERROR: failed to mount smb share "//192.168.0.200/CAM/iphone" at "/media/motioneye_192_168_0_200_cam_iphone_kjones"
   ERROR: access to network share //192.168.0.200/CAM/iphone failed: cannot mount network share
   ERROR: mjpg client timed out receiving data for camera 1 on port 9081

Is this a file permission issue on the MotionEye add-in side?  I am not very "Linux-savvy" and am just starting to dip my toes into all of this... Any help would be super appreciated!

starbasessd, 2024-05-23:
I would suggest to not run the motionEye plug in from HA.
I would suggest installing motionEye 0.42 or 0.43 on another OS, that way it can be supported without dealing with HA which has always had issues.

starbasessd, 2024-05-23:
I use VirtualBox 7.x for one host, running Debian 10 Buster lite 64  bit, 4GB allocated ( but never more the 1GB used), 48GB local storage, but could set to 10-15GB, currently 3GB used), and a 14TB NAS for video and pics. I will be upgrading to Deb 12 there soon. Running mEye 0.42.1, motion 4.3.2, 6 cameras, 2 USB and 4 IP.
I use a Xen server for my 2nd host, and allocate closer to my lower standard, running Deb12, 2GB, 20GB, and a connection to the 14TB NAS. Running motionEye 0.43, latest motion. Same IP cameras, and shared video from first motionEye for the USB cameras.
Both do the job, about 15fps across all cameras.
I do recommend running lite vs full desktop...

kevo970, 2024-05-26:
So I was googling my error and finally found out the issue was with my VM config on trueNAS.

I didn't have the NIC setup as bridge so my entire HA  installation couldn't reach my NAS. Once I setup the bridge everything is working as well as I need it to, so I'm gonna keep running with the HA add-on for now, but figured I'd update the thread in case anyone bumps into this later.