motioneye-project/motioneyeos #3033: RPi Zero W doesn't get an IP on first boot

Cubytus, 2024-05-25:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: dev20201026

### Board Model

I am using the following board/model: Raspberry Pi Zero W.

### Camera

I am using the following type of camera: can't know as the board doesn't get an IP address.

My camera model is: Official Raspberry Pi camera.

### Network Connection

My motionEyeOS unit is connected to the network via: **USB-Ethernet adapter**

### Peripherals

I am using the following peripherals that I consider relevant to this issue: For the sake of this test, I added a few connections that aren't necessary in a normal setting

 * mini-HDMI to VGA adapter connected to a standard monitor
 * USB OTG adapter (official RPi model)
 * USB 3.0 hub
 * wireless keyboard

### Log Files

None as I have no LAN connection to this board and no user/password pair
 
The basics being said, I wanted to reinstall MotionEyeOS from scratch. Even if development is frozen, I assumed it should be working reasonably well in a LAN. I followed the wiki's instruction to **image a microSD card** with the latest dev20201026 version, inserted it in the board, made all connections (Ethernet included) then powered it up. 

It stops at the "Welcome to meye-12345678!" "If you want to see a friendly user interface, point your browser to the IP address of this board"
Of course, it doesn't display what this IP address might be, and all I get is the one-line message showing **the date and CPU temperature**.

The date itself isn't updated despite Ethernet being connected (as shown by both its link/act lights flashing)

A quick LAN scan with Angry IP scanner doesn't display any new device and what's even weirder, a look at the admin page for the router doesn't display the MAC of the USB-Ethernet adapter, which is known to be working.

Now it seems that many issues are happening at the same time, but I can't find out where to start.

starbasessd, 2024-05-25:
How did you image? If you RPiImager, try using dd, or Win32DiskImager, or Etcher, as RPiImagewr corrupts it due to it doesn't handle the 3 partitions and auto-creating the extra info.
Either wifi or ethernet is pulling an IP address because if not, it would go into a panic and reboot loop.
If you have a keyboard and monitor connected, you can access the logs, insert a USB memory key and save the logs there. Default user is root or admin with no password. The command 'ip a' will report the ip addresses of the usb ethernet or on board wifi.
I used Win32DiskImager, PiZeroW, Makerspot USB OTG hub, Belkin USB Ethernet adapter, and my 'standard' wifi wpa_supplicant.conf file.
Booted normally.
Last few lines on screen were:
# Interface eth0 has IP address 192.168.2.xxx/24
 # Interface wlan0 has IP address 192.168.2.yyy/24
 # Default gateway is 192.168.2.zzz
 # DNS server address is 192.168.2.ddd
logs are in /data/log/

starbasessd, 2024-05-25:
standard format of my wpa_supplicant.conf file:
= = = = =
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
    ssid="ssid"
    psk=password
}
= = = = =

Cubytus, 2024-05-25:
I used **balena Etcher** for imaging. On boot, there are quite a bit of messages about a **read-only file system**, which shouldn't happen.

Wifi isn't configured at all so can't pull an IP. But how would Ethernet pull one yet not show in the router's interface?

I tried to log in, but one thing is sure, pairs

- admin/(empty password)
- root/(empty password)

aren't the default ones.

starbasessd, 2024-05-25:
normal /data/log/boot.log (only last boot for brevity)
= = = = =
---- booting motionEyeOS dev20201026 ----
 * Detecting disk device: /dev/mmcblk0
 * Checking data filesystem: done
 * Mounting filesystems: done
 * Mounting overlay filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Configuring CPU frequency: done
 * Starting syslogd: done
 * Loading device-tree overlays: done
 * Starting throttle watcher: done
 * Starting eudev: done
 * Starting watchdog: done
 * Starting rngd: done
 * Preparing AP interfaces: ap0
 * Starting wpa_supplicant: done (brcmfmac)
 * Configuring wireless network: dhcp
 * Setting current date using SNTP: Sat May 25 19:08:06 UTC 2024
 * Starting ntpd: done
 * Starting crond: done
 * Starting sshd: done
 * Starting proftpd: done
 * Starting smbd: done
 * Starting nmbd: done
 * Starting motioneye: done
 * Interface eth0 has IP address 192.168.2.xxx/24
 * Interface wlan0 has IP address 192.168.2.yyy/24
 * Default gateway is 192.168.2.zzz
 * DNS server address is 192.168.2.ddd
= = = = =
If you are getting 'read only' filesystem errors during boot, something is corrupt. I would try aq fresh image.

Littlechay, 2024-05-27:
Angry IP scanner will only scan a couple of ports by default. Make sure you
have it scan the correct ports.

starbasessd, 2024-05-27:
@Littlechay by default, Angry searches the port 80 default port for motionEyeOS. If he were looking for another OS running motionEye, then he'd need to search port 8765.