motioneye-project/motioneyeos #3034: Self Access Point: Rebooting because no network

rdb9879, 2024-05-26:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: dev20201026

### Board Model

I am using the following board/model: Raspberry PI 3B+

### Camera

I am using the following type of camera: Simple MJPEG Camera.

My camera model is: Raspberry Pi Camera Rev 1.3

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

I am trying to do a headless install to generate its own WiFi Access Point (no wired or other WiFi connections). I've copied the hostapd.conf and dnsmasq.conf verbatim from [https://github.com/motioneye-project/motioneyeos/wiki/WiFi-Access-Point](url) and changed "wlan0" to "ap0" per the instructions. It ends up in an infinite boot loop complaining that there is no wired or wireless connection. I tried creating a wpa_supplicant.conf file to connect to its own ssid using the ssid and passphrase from hostapd.conf, but I get "failed (brcmfmac)." 

What are the correct steps to install MotionEyeOS in headless mode, to create its own Access Point? I will ultimately have a tablet connect to its generated WiFi network and view the fast network cam in a web browser.

starbasessd, 2024-05-26:
Did you also add the files: [static_ip.conf](https://github.com/ccrisan/thingos/wiki/static_ip.conf) and [os.conf](https://github.com/ccrisan/thingos/wiki/os.conf) also?

rdb9879, 2024-05-26:
I did not. Do you have an example file that would be relevant for Access Point?

starbasessd, 2024-05-26:
[AP files.zip](https://github.com/motioneye-project/motioneyeos/files/15448649/AP.files.zip)

rdb9879, 2024-05-26:
Thanks for the files. Just so I'm clear: I do or do not need to include wpa_supplicant.conf?

starbasessd, 2024-05-26:
You do not need wpa_supplicant.conf if you pre-load the files I gave (edit as needed) in the /boot partition before first boot.

rdb9879, 2024-05-26:
Reburned a new image, and before ejecting the SD card, I put your exact files on but it's still stuck in a reboot loop. I'm seeing:

`Preparing AP interfaces: ap0
Starting hostapd: done (brcmfmac)
Configuring wired network: no link
panic: rebooting in 20 seconds (caused by network)`

Any other suggetions?

starbasessd, 2024-05-26:
What base image are you using, and how are you burning it to the SDCard?
This [image](https://github.com/motioneye-project/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi3-dev20201026.img.xz)?
Do not use RPi Imager, use Win32Disk Imager (preferred Windows) or Balena Etcher, or dd.

starbasessd, 2024-05-26:
I just tested, and it works on a Pi3B on a 16GB SDCard, using the Win32DiskImager, and the files above.

rdb9879, 2024-05-26:
Yes, that was the exact image I used, and I did indeed use Win32Disk Imager, and I unzipped your AP Files.zip and copy and pasted them onto the burnt image. I'm going to try a different RPI tomorrow, maybe the WiFi chip has died (I've had other RPI 3's have their WiFi chips die).

starbasessd, 2024-05-26:
Does it work with a regular image (Raspbian 10, 11, or 12) and your wpa_supplicant.conf file to connect to your home wifi??? If yes, then it's not the wifi chip.

rdb9879, 2024-05-27:
I rolled back to the version I had the most experience with (20190119) and managed to get the AP working again. So it's not a hardware issue, and I have no idea why you were successful with dev20201026 and I was not, being I copied and pasted your exact files. Did you have a physical Ethernet Cable plugged in? (I did not)

starbasessd, 2024-05-27:
@rdb9879 no, just the standard wifi, no ethernet.

rdb9879, 2024-05-27:
Well I have it working with the older version so I need to move on with my project. Probably won't be debugging anymore at this time. Thanks for your help