motioneye-project/motioneyeos #3036: friendly user interface problem

jeevank88, 2024-06-07:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 
[image]

i am getting this error please help me 

my wpa_supplicant files is 

ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=IN

network={
ssid="robot"
psk="12345678"
}

starbasessd, 2024-06-07:
What board and what version of motionEyeOS are you using?

jeevank88, 2024-06-07:
> What board and what version of motionEyeOS are you using?
Board = raspberry pi zero w
Version = 20200606(kernel 4.19)

jeevank88, 2024-06-07:
> > What board and what version of motionEyeOS are you using?
> > Board = raspberry pi zero w
> > Version = 20200606(kernel 4.19)

jeevank88, 2024-06-07:
> What board and what version of motionEyeOS are you using?

Board = raspberry pi zero w
Version = 20200606(kernel 4.19)
For network I am using mobile hotspot

jeevank88, 2024-06-09:
> What board and what version of motionEyeOS are you using?

Hello bro please reply

starbasessd, 2024-06-09:
Please use the version here: [dev20201026](https://github.com/motioneye-project/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi-dev20201026.img.xz)
Be sure to place the wpa_supplicant.conf file from above in the first (boot) partition with an ssh or ssh.txt file.
It looks like you may have tried to upgrade an earlier version or that the SDCard got corrupted.

jeevank88, 2024-06-09:
> Please use the version here: [dev20201026](https://github.com/motioneye-project/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi-dev20201026.img.xz)
> Be sure to place the [...]

I tried bro in first boot it's coming like this [image]

Even I have added ssh file also

starbasessd, 2024-06-09:
Wrong version. You are trying 20200606 not dev20201026.
How are you creating the SDCard? Use Win32Disk Imager (Windows), or Balena Etcher (Windows, MacOS or Linux) or dd (MacOS, Linux, or RaspiOS).
That last screenshot is a normal first boot, but the wrong version.

jeevank88, 2024-06-09:
> Wrong version. You are trying 20200606 not dev20201026.
> How are you creating the SDCard? Use Win32Disk Imager (Windows), or Balena Etcher (Windows, MacOS or Linux) or dd (MacOS, Linux, or RaspiOS).
> That last screenshot is a normal first boot, but the wrong version.

In both versions it's coming like that
[image]

starbasessd, 2024-06-09:
Where is the wpa_supplicant.conf file?
Since I am not there, I ask a number of questions to help me figure out what is going on.
The first screenshot you posted showed a normal boot and normal running, but without the ip address, and the right hand side of the screen cut off.
It stated that it wasn't abler to find a file, and that a file it expected to be able to write to was in a read-only partition. This usually indicates a corrupted image, or bad SDCard. Most common coruption is using RPiImager, which does not handle 3 partition images like motionEyeOS, especially if you try to use it to pre-set the wifi settings.
Further posts indicated that you were running a known 'iffy' version, and I said you needed to use dev20201026, and add the wpa_supplicant.conf file prior to first boot, and recommended several apps to create the SDCard.
I have the hardware here to reproduce most RPi issues, PiZeroW to Pi4 4GB & 8GB models. I set a PiZeroW up with dev20201026 and a wpa_supplicant.conf file to talk to my phone's hotspot, and was able to c0onnect from the chrome browser on the phone to it no issue.

jeevank88, 2024-06-09:
> Where is the wpa_supplicant.conf file?
> Since I am not there, I ask a number of questions to help me figure out what is going on.
> The first screenshot you posted showed a normal boot and normal [...]

Later I have added wpa_supplicant file
[image]

I think there is problem in sd card because so many times I have flashed
I will try in new sd card

starbasessd, 2024-06-09:
How big? I've had issues with SDCards over 128GB. Takes a long time even at 64GB.

jeevank88, 2024-06-09:
> How big? I've had issues with SDCards over 128GB. Takes a long time even at 64GB.

I was using 16gb sd card

closed 2024-06-10 by jeevank88