motioneye-project/motioneyeos #3038: No video output, no WiFi connection, no log files

Ahndalu, 2024-06-22:
### Preliminary Docs

Yes, I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

Yes, I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (motioneyeos-raspberrypi3-20200606.img.xz)).

### Board Model

I am using the following board/model: (Raspberry PI 3A+).

### Camera

I am using the following type of camera: (Raspberry Pi Camera Module NoIR 3 Wide, connects
to Pi by ribbon cable ).

My camera model is: (it has a 1.14 on it).

### Network Connection

My motionEyeOS unit is connected to the network via: (attempting to connect to WiFi. It has
failed. It has never connected successfully.  This model of Pi has no wired ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * Nothing relevant.  
 * I have a wired USB keyboard plugged into the board that isn't yet useful.
 * I have a regular HDMI monitor plugged into board
 * I have power supply plugged into board
 * I have the camera ribbon cable plugged into the camera ribbon cable connector on board
 * That's it

### Log Files

I consider the following log files relevant to this issue:

 * /var/log is empty.  No files at all
 
[image]

I have created the image on SD card several times.  I have created and place in the boot partition
the wpa_supplicant.conf file shown in screen shot. I created wpa_supplicant.conf and placed it
in /boot prior the first boot of the card on the pi.  I placed copies of wpa_supplicant.conf into
/boot into /data/etc and into /etc  (I wasn't sure I was supposed to do that or not).  I have tried many variations of the wpa_supplicant.conf syntax.  While the red LED goes on, on the Pi, and the green LED blinks a lot,
I get no video output to a monitor, and I get no connection to the WiFi network. I verified that I can connect
to the WiFi network using same credentials with a laptop.  I am a newbie.  I have tried a large number
of things to make it work, reading docs and trouble reports.  The SD is 256GB.  The board is stamped 2022.
There is nothing in the /var/log directory - no files at all, zero. Prior to installing MotionEye, I had the conventional Rapsberry Pi OS on the board and was able to view video with the camera.
Don't know what to do next.

Ahndalu, 2024-06-22:
only thing in /data is what I created and put there.  (I am accessing files by pulling the card out of the pi and sticking it into a laptop to view)  /etc/network contains only directories, no files.

starbasessd, 2024-06-22:
DO NOT USE RPiImager! Use Win32DiskImager, Balena Etcher, or dd. RPiImager does NOT support 3 partition images like motionEyeOS, and does not work with mods made by using the presets created by RPiImager.
[There are good instructions here to get a good image.](https://github.com/motioneye-project/motioneyeos/wiki/Installation)
Using a 256 GB drive will take a long time (up to several minutes) for first boot, as it will take a long time to expand and format the data partition.

starbasessd, 2024-06-22:
Also, [the max standard size for the Pi3s is 32GB](https://www.google.com/search?client=firefox-b-1-d&q=max+sd+card+size+rpi+3a%2B) unless you expand the partition and format it in another machine first.  And motionEyeOS doesn't support that.

Ahndalu, 2024-06-22:
Thank you for these tips. I will have to go buy either 32GB or 64GB cards.   64GB is too big?

starbasessd, 2024-06-22:
I don't have a 3a+ to test. Works on a 3B.

Ahndalu, 2024-06-22:
Trying a 64GB.  Do I need to put wpa_supplicant.conf  in /etc   /data/etc   and /boot
or just in /boot partition?

Ahndalu, 2024-06-22:
Sorry, typo.  I mean:
Do I need to put wpa_supplicant.conf in 

/etc 

and

/data/etc 

and 

/boot

or just in 

/boot 

partition?

starbasessd, 2024-06-22:
just /boot prior to first power on

Ahndalu, 2024-06-22:
I used gnome-disk-utility 42.0
UDisks 2.9.4 (built against 2.9.4)
in linux ubuntu,
screen shot attached,
to write the image to a 64GB SD card. I placed wpa_supplicant.conf
in the boot partition in  /boot 
and
/etc in main partition and
/data   in main partition.
It didn't work.  Results were the same.
Please let me know the next step?   
Thank you
![image](https://github.com/motioneye-project/motioneyeos/assets/173562645/e115c3dd-856d-40a7-9606-49def54be127/

starbasessd, 2024-06-23:
Try [this image](https://github.com/motioneye-project/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi3-dev20201026.img.xz)
Only place the wpa_supplicant file in the /boot partition. Do not touch the other 2 partitions.

Ahndalu, 2024-09-04:
Sorry for being so long in continuing this thread.   I got 32GB cards because I had
learned that the 64GB cards I was using for Rapsberry Pi 3 Model A+ probably would not
work.  I wrote the image linked below in the "this image" link to a 32GB card.  I placed a
wpa_supplicant.conf file in the 31MB boot partition.  When I turn on the Pi, I get red LED and then blinking green LED.  However, there is no output on the HDMI.  The Wi-Fi router never gets a DHCP connection with the Pi.  I can connect a different laptop to the Wi-Fi router via DHCP just fine, and can access the router's administration pages and see that the laptop is the only device connected to the router over the Wi-Fi, and the laptop received an IP address fine.

When using the image prior the one in this post, I tried a number of things in the
config.txt in the boot partition. I tried giving a static IP.  There were never any changes
in behavior.   /var/log in the non-boot partition is an empty directory (after having booted
the pi, and then looked at the sd-card on another machine)

I do not know what to do next.  Please advise?  Thank you.

Here is my wpa_supplicant.conf file  (with real password and network replaced)

country=US
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1

network={
    ssid="MyNetwork"
    psk="MyPassword"
    key_mgmt=WPA-PSK

Ahndalu, 2024-09-05:
And a second (new) attempt was made:   I burned a new image on 32GB SD card using
 ./writeimage.sh -d /dev/mmcblk0 -i "/path/to/motioneyeos-raspberrypi3-dev20201026.img" -n 'yournet:yourkey' 

The script executed fine.  It gave output messages that look correct, and told me I could eject the SD card.  I put the SD card into the Pi and turned it on.  

The same symptoms occur:  no HDMI video.  Red LED lit.  Green LED blinking.  /var/log remains empty.

I confirmed that 'yournet' and 'yourkey' values for my setup had no syntax errors as my laptop can connect to the Wi-Fi router using those credentials and I copied & pasted them rather than typing them. 

Does MotionEyeOS just not work with a Pi A 3+ ?

starbasessd, 2024-09-06:
Theoretically, the mEOS image for the 3B+ should work on the 3A. The only real difference should be the [memory (1GB vs 512MB on the 3A)](https://www.digikey.com/en/maker/blogs/2019/raspberry-pi-3-a-vs-b), the lack of USB and Ethernet should not be an issue. I personally do not have a 3A+ for testing. I would (and do) recommend running RPiOS Buster or Bullseye Lite versions and installing motionEye 0.42 (requires python 2.7) from [here](https://github.com/motioneye-project/motioneye/wiki/Install-On-Raspbian) or Bookworm and 0.43 (requires python 3.x) from [here](https://github.com/motioneye-project/motioneye/releases)

Ahndalu, 2024-09-07:
I have installed Raspberry Pi Lite, version 12 bookworm now.  I have tried to install MotionEye. It produces errors when I launch it, post install.  (it fails to find the setup.py even though all the paths seem to be correct)   I am using MotionEye version .42.1   I have python version 3.11    Does this mean I need to upgrade  to version .43?   The note on the .43 page says "motionEye requires now Python 3.7 or later" - is python 3.11 ok?   Given my setup, what is best/fastest thing to change?

Ahndalu, 2024-09-07:
I upgraded to the pre-release .43.  I had to upgrade the tornado version.  Then I finally got MotionEye to start without errors.  Thank you for your help.