motioneye-project/motioneyeos #3040: USB camera not even registering with motioneyeos "add camera" interface

rcho5840, 2024-06-24:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606

### Board Model

I am using the following board/model: Raspberry Pi 4B

### Camera

I am using the following type of camera:V4L2

My camera model is: Emeet C960 / Testing on various other usb cameras

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

N/A? Usb camera setup

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

starbasessd, 2024-06-24:
If you are using a non-default (included with motionEyeOS) start4.elf or other kernel files, USB doesn't work, and it's a known issue.
There are only a couple of bootloaders (firmware) that work on the Pi4, and they are dated around 4/2020. You can search here for how to check and flash other bootloaders, but it seldom works going from the most current back to one of the older ones.
Your best bet would be to load RaspiOS Buster or Bullseye Lite and install motioneye 0.42.1 for Python2.7, or you can try Bookworm with Python 3 and motioneye 0.43.0.

rcho5840, 2024-06-24:
@starbasessd are there similar issues on the pi3? pi zero 2W? I'm getting frusturated and wondering if it might just be worth trying on these boards

starbasessd, 2024-06-24:
MotioneyeOS (dev20201026) works fine without mods. The forked PiZero2W fork works fine, too... https://github.com/motioneye-project/motioneyeos/issues/2875

rcho5840, 2024-06-25:
@starbasessd I also tried 20201026 on the pi4 and it said that start.elf is not compatible. Any thoughts? I tried replacing start.elf and fixup.dat with the same files of the newest debian bookworm but it still didn't work

starbasessd, 2024-06-25:
If you are using a non-default (included with motionEyeOS) start4.elf or other kernel files, USB doesn't work, and it's a known issue.
There are only a couple of bootloaders (firmware) that work on the Pi4, and they are dated around 4/2020. You can search here for how to check and flash other bootloaders, but it seldom works going from the most current back to one of the older ones.
The bootloader system is unique to the Pi4s and Pi5s, and you can check the bootloader in the eeprom on them by running the command:
vcgencmd bootloader_version

one of mine reports:
pi@raspi1:~ $ vcgencmd bootloader_version
2024/04/15 14:12:14
version e608a69d210bb737ebb095ad04f7f1a05373a6ec (release)
timestamp 1713186734
update-time 1715711612
capabilities 0x0000007f

and doesn't run motionEyeOS dev20201026, but another that reports the date of 2020/04/16 works.
Downgradinmg seldom works though.

rcho5840, 2024-06-25:
so would the pi3 and dev 20201026 probably work?

starbasessd, 2024-06-25:
it does with a Tiga USB Dashcam quite well on a Pi3 and motionEyeOS dev20201026

starbasessd, 2024-06-25:
Other options are:
load RaspiOS Buster or Bullseye Lite and install motioneye 0.42.1 for Python2.7, [instructions here](https://github.com/motioneye-project/motioneye/wiki/Installation)
or you can try Bookworm with Python 3 and motioneye 0.43.0. [instructions here](https://github.com/motioneye-project/motioneye#installation)