motioneye-project/motioneyeos #3053: No way to get network work in a Pi 3B+

marciano1, 2024-12-29:
motionEyeOS version: motioneyeos-raspberrypi3-20200606.img
Raspberry PI 3B+).
NoIR camera, extra IR light.
My motionEyeOS unit is connected to the network via: WiFi, Ethernet.

From Ubuntu and Raspberry Pi Imager v1.7.2 (also tried with Balena Etcher) I wrote a microSD card.
Boot from Rpi 3B+ doesn't get network connection, checked with `sudo nmap -sn 192.168.1.0/24`
So I tried an alternative from Wiki
`sudo sh writeimage.sh -d /dev/sdf -i "path-to/motioneyeos-raspberrypi3-20200606.img" -s "192.168.1.104/24:192.168.1.251:8.8.8.8"`
(static IP, Gateway, DNS)

No way to get it work, no IP available. Tried changing the RJ and power cables.
Network is okay in this rPI with other OS like Bookworm.
The strange thing to me is that turning on rPi with motionEye green led on board continuously flashes slowly 4 times and then rapidly 7 times.
  
I have another motionEyeOS in a Pi 4 and both Ethernet and WiFi are discoverable.
I don't get what else to try.

dermotx, 2025-01-28:
I've the same problem. Did you manage to fix it?

bbillp, 2025-01-30:
Before the first bootup connect via CAT5/6/xx, in raspi-config make sure USA or your country is the WiFi country and set the SSID.  Maybe a step was missed when burning the midnight oil.

dermotx, 2025-02-01:
Thanks for that but it didn't work either.

ivorgetnow, 2025-02-27:
I have also seen this problem, and can offer only a partial bit of progress since I am not a native propellerhead.

I use motioneyeos on a raspberry pi 3B+. Or I did until the pi's ethernet port failed. Wifi still works but it's physically out of range - it's fitted into a bird nestbox and survived several years. It's POE with a third party hat. I wanted to stay with a 3B+ as it's less power hungry than a pi4. Found a replacement at microcenter, yay. Flashed a new SD and - 4 long, 7 short LED blinks, no boot.

Search begins. This blink pattern reports a power fail type B. Turns out that my 3B+ is a newer build and breaks MEYEOS since it is no longer maintained. Still faintly hopeful, I found a pdf on the rpi docs pages discussing firmware updates. There is a link in the pdf to obtain a firmware-stable.zip from which can be extracted replacements to specific files. 

What to do? Extract the zip; go into boot folder, find start.elf and fixup.dat files. Copy these into the MEYE boot SD replacing the ones from the fresh flash of 20200606, the latest (last ever?) release of MEYE. 

After this I did get booted into MEYE. This was the only victory however, because when I attempted to add/configure my MMAL camera, MEYE errored out, and since then I cannot change, add, or remove the camera configuration.

So hey, if anyone can carry the ball a little further I'd love to hear. I'm out of answers.