motioneye-project/motioneyeos #3055: mE on proxmox - wtf ...

dummbold, 2025-01-11:
### Preliminary Docs

**X** I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

**X** I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

**?** I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

**Mac/64 w. Proxmox -** I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

**MJPG -** I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

**Edimax IC-3115W -** My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

**LAN** - My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

N  * (e.g. USB WiFi adapter)
N  (e.g. External Hard Disk) 

### Log Files

I consider the following log files relevant to this issue:

N  * (attach e.g. motioneye.log)
N  * (attach e.g. boot.log)
 

I am running Proxmox on a Mac/64 with several VMs and Containers (eg. HAOS). These days I tried to connect a 10yr old Webcam to HA. It was a complicated ride, because there is no full documentation. So I found mE and installed the Container Version that is provided via the "proxmox helper scripts".

In the installation script is a link saying that they got it from this ressource, but I can't find any documentation for the Container-Version. In my version a lot of stuff is missing (eg. motion detection, video and photo recording - see attached screenshot.

And now?

[image]

OberfragGER, 2025-02-24:
Are you using LXC or docker? I had similiar problems with the Helper Scripts (i think a time ago) and decided to install it by myself. Within the last days i found the docker copose. Works like a charm under Proxmox - docker.

Dont stick with that. I dont know whats wrong there. Install manually or use docker.

dummbold, 2025-02-25:
Meanwhile I changed from a direct VM on Proxmox to an installation inside Homeassistant. The problem worsened!  I use the Integration "MJPEG IP Camera" to bring the camera into HA - and this works fine, but without any additional functions that I require. In the "motioneye" Add-On (server) I can see the camera picture - but still without the functions. The integration recognizes the camera - but it is "not available" ...